julianduque/beaglebone-io #15: Update i2c support for new methods

MrYsLab, 2015-01-01:
The new i2c methods specified at https://github.com/rwaldron/johnny-five/wiki/IO-Plugins need to be implemented.

julianduque, 2015-01-01:
@MrYsLab so all the sendI2C* will be deprecated?

MrYsLab, 2015-01-01:
Were it so easy ;-). Some methods are "overloaded", so I will need to add the logic in the library code to handle the overloading and pull out the arguments correctly. I am not sure why this complication was added, but I will handle it.

I am beginning to feel like Sisyphus. Just when I get to the top of hill ... :-).

I want to map the new methods with their overloaded signatures to the i2c-bus methods and have @fivdi  check out if my understanding is correct before coding.

BTW, we also need to perform an emit for the read methods. Not a big deal, but just wanted to let you know.

I will create another i2c branch. Since we used that name before, will that cause any problems? I can easily use a unique name, like i2c-1.

fivdi, 2015-01-01:
Here's the method mappings from IO Plugins to i2c-bus:

IO Plugin Method | i2c-bus Method |
--- | --- |
i2cWrite(address, inBytes) | i2cWriteSync(addr, length, buffer) |
i2cWrite(address, register, inBytes) | writeI2cBlockSync(addr, cmd, length, buffer) |
i2cWriteReg(address, register, value) | writeByteSync(addr, cmd, byte) |
i2cRead(address, register, bytesToRead, handler) | readI2cBlock(addr, cmd, length, buffer, cb) |
i2cRead(address, bytesToRead, handler) | i2cRead(addr, length, buffer, cb) |
i2cReadOnce(address, register, bytesToRead, handler) | readI2cBlock(addr, cmd, length, buffer, cb) |
i2cReadOnce(address, bytesToRead, handler) | i2cRead(addr, length, buffer, cb) |

And the mappings for some argument names:

IO Plugin Argument Name| i2c-bus Argument Name |
--- | --- |
address | addr |
register | cmd |

IO Plugin methods use arrays, i2c-bus methods use buffers.

i2c-bus methods that have a buffer argument also have a length argument (similar to fs.read|write[Sync].)

MrYsLab, 2015-01-02:
@fivdi  Wow!, what a gift. I just sat down to map things out, and your mappings appeared. Thanks.

julianduque, 2015-01-02:
@MrYsLab create a `refactor-i2c` branch ;), I would love to give you a hand on this !

MrYsLab, 2015-01-02:
@julianduque Thanks. Let me know your thoughts about breaking up the work. The lib code looks pretty straight forward. The [Firmata](https://github.com/jgautier/firmata) code illustrates how to handle  overloaded methods. I did not fully understand your comments on returning the PIN mode of UNKNOWN - "you can set the pin mode as an internal state not need to interact directly with the bbb". I am not sure how this is implemented. I am also unclear on the comments in the WikI about unknown having a value of 99. In j5/lib/board.js UNKNOWN is a string. In Firmata, UNKOWN is defined as 0x10, so I I am not sure what 99 is. I will fire off a question to Rick.

@fivedi did a great job in mapping the calls to his library, so that will make things much easier.

I will create the branch and populate the file with the signatures for the new methods. I can then do a pull request so you can work on implementing any of the methods if this what you would like to do or if you wish, you can use the signatures to create the tests.

Also, if you would like to take the whole thing on, I have no problems standing in the wings. Just another option.

julianduque, 2015-01-02:
@MrYsLab go ahead and push directly to the branch you created, I'll review it and jump into it, I can do the tests though.

Regarding the pin mode the idea is to setup the pin mode in the internal state (https://github.com/julianduque/beaglebone-io/blob/master/lib/beaglebone.js#L217) but not calling it on the BBB because there is no support for it as you said (https://github.com/julianduque/beaglebone-io/blob/master/lib/beaglebone.js#L226), for the UNKNOWN string vs hex I would like to summon @rwaldron to solve this question.

MrYsLab, 2015-01-02:
@julianduque Rick sent me this on gitter, so I think we can safely ignore it.

"as it turns out, on the IO Plugin side, the UNKNOWN pin mode is unnecessary. I forgot to take it out

julianduque, 2015-01-02:
@MrYsLab awesome then!

MrYsLab, 2015-01-02:
@julianduque I just pushed lib/beaglebone.js to the refactor-i2c branch. It contains the signatures for the refactored i2c methods, but no implementation yet.

julianduque, 2015-01-02:
@MrYsLab awesome!! I'll jump on it tonight ;)

fivdi, 2015-01-03:
@julianduque  a requirement for the i2cRead methods is to read continuously and asynchronously. Perhaps you know what this means, but I didn't, so I took a look at galileo-io to see. The [delay](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L412) argument passed to i2cConfig is used by [i2cRead](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L537) to determine the read rate. [galileo.js](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js) will be very helpful while refactoring.

MrYsLab, 2015-01-03:
@fivdi @julianduque Brian - great detective work! I think the best implementation approach I can take, is to port as much of the Galileo i2c code as possible. Rick has some experimental stuff commented out, which tells me that more changes are on the way. By staying as close to Galileo as possible, perhaps we can avoid major future rewrites and not have to be contortionists to add new features as they come along.

fivdi, 2015-01-03:
Some things need to be done differently. For example, because the mraa code is synchronous, [this line of code](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L529) runs in the same thread as the main event loop and blocks that thread. The equivalent i2c-bus method is asynchronous and will run in a different thread and will not block the event loop thread. This means that calling [setInterval(..., delay](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L526-L537) with a delay of 0 (the default) will not work. Let me know if this is not clear an I'll explain it in more detail.

MrYsLab, 2015-01-03:
@fivdi If the event is emitted in the callback for the i2c-bus read method wouldn't that work?

rwaldron, 2015-01-03:
That experimental stuff won't affect the operational semantics of the API. 

> For example, because the mraa code is synchronous, this line of code runs in the same thread as the main event loop and blocks that thread.

Yep. But it's safe to assume this is **wrong**—if you can do reads asynchronously, then do that. The resources working on the platform at Intel aren't interested in making the JS bindings behave in a way that works correctly for Node.js. The problem is that they're writing the source in C and then auto-generating the JS bindings with Swig—which results in this process blocking abomination. 

@fivdi if you were to support Galileo and Edison with https://github.com/fivdi/i2c-bus, we would _absolutely_ use that instead (_hint, hint_)

fivdi, 2015-01-03:
@MrYsLab Yes, the event can be emitted in the callback for the i2c-bus read method. However, setInterval shouldn't be used for continuous reading.

Instead of using setInterval like this (as galileo-io does):

```js
    setInterval(function() {
      _i2c.readI2cBlock|i2cRead(..., function (...) {
        // emit event here
      });
    }, delay);
```
setTimeout should be used instead:
```js
    function readLoop() {
      _i2c.readI2cBlock|i2cRead(..., function (...) {
        // emit event here
        setTimeout(readLoop, delay);
      });
    }
    setTimeout(readLoop, delay);
```
Although these look very similar, they can be very different, especially when delay has low value, like 0, the default.

If delay is 0, the setInterval variant will result in _i2c.readI2cBlock|i2cRead being called on every turn of the event loop. The code will not wait for the previous call to complete (remember it's running in a different thread). The setTimeout variant will prevent this from happening.

galileo-io can use setInterval because the mraa code is synchronous and runs in the same thread.

@rwaldron support Galileo and Edison with i2c-bus? hmm, why not. I'll have to order some boards first.

MrYsLab, 2015-01-03:
Makes sense. I hope to have the coding done tomorrow sometime, and if I have time will test it out on hardware. If all works, will push it to the repo.

rwaldron, 2015-01-03:
@MrYsLab don't bother with Galileo Gen 1. Gen 2 and Edison are the only ones we care about moving forward. 

Also, I will fix Galileo-IO to use setTimeout instead.

MrYsLab, 2015-01-03:
@rwaldron I am basing the BBB code on https://github.com/rwaldron/galileo-io/tree/master/lib. I am assuming that this is the code for Gen2 and Edison. Let me know if I should be looking elsewhere.

rwaldron, 2015-01-03:
@MrYsLab Yep. I was just replying to @fivdi here: 

> I'll have to order some boards first.

MrYsLab, 2015-01-03:
@fivdi I completed the code sooner than I thought. Correcting a previous statement, I am pushing out to the repo for review even though I have not connected it to the hardware yet. I should be testing it in the morning, but I don't want you to feel that I am pressing for an overnight review. Take your time.

fivdi, 2015-01-04:
@MrYsLab I'm traveling most of the day today so won't get around to looking closely at it till tomorrow. I did see two things and added comments to the commit. I see that the implementation is very close to that of Galileo-IO so it doesn't use _i2c.readI2cBlock.

An additional point is that because the i2c-bus asynchronous methods are being used, continuously reading from two devices concurrently will mess things up as there will be two threads using the bus concurrently, i.e., the two calls to _i2c.i2cRead will result it two threads using the bus concurrently. This will work in Galileo-IO as everything is synchronous.

On the other hand, I don't think Galileo-IO supports reading continuously from two devices either. The call for the second device will mess up the address for the first device [here](https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L517), I think.

MrYsLab, 2015-01-04:
@fivdi I incorporated your change requests and made some other changes from the previous version. I tested with HMC5883L using the Magnetometer demo (it calls compass.js), which uses the latest i2c calls. I also tested with 8x8 bi-color LED matrix, which still uses the old style and both work.
Code is available in the refactor-i2c branch.

rwaldron, 2015-01-04:
@fivdi yikes, I'll test that today (in Galileo-IO)

rwaldron, 2015-03-23:
@julianduque what's the status on this? I can pitch in if you need more hands

julianduque, 2015-03-23:
@rwaldron definitely, I'm out of bandwidth these week

julianduque, 2015-07-24:
Landed on `beaglebone-io@1.2.0`

closed 2015-07-24 by julianduque