julianduque/beaglebone-io #18: Unable to run script example/led.js

vallettea, 2015-02-22:
Hello, thx for this module.

I get a error: 
```
error: Unable to find LED: usr3
error: Unable to find gpio: /sys/class/leds/beaglebone::usr3/brightness
error: Unable to write to /sys/class/leds/beaglebone::usr3/brightness
```

amI missing something ? Is there a requirement with the kernel or whatsoever ?

fivdi, 2015-02-22:
It looks like you're running an OS without the required device tree support. Perhaps a very old Ångström distribution or maybe an up-to-date Ubuntu distribution?

Does the directory `/sys/devices/bone_capemgr.n` (where n is a number) exist on your system?

If it doesn't exist, the required device tree support isn't there.
It's best to use the recommended version Debian. See [latest images](http://beagleboard.org/latest-images).

vallettea, 2015-02-22:
yes I'm running ubuntu and there is no such a directory. You should specify that it requires debian in the readme for later users.
I'm testing it now with debian.
thx

fivdi, 2015-02-22:
Yes, you're quite right. Apologies for any inconvenience caused. A note has been added to the [Install](https://github.com/julianduque/beaglebone-io#install) section of the readme to help avoid similar problems for others.

vallettea, 2015-02-22:
no pb ;) and thanks again for the lib. 
I'm thinking of writing the equivalent for the odroid c1. Do you have any recommandations ?

closed 2015-02-22 by vallettea

fivdi, 2015-02-22:
Do you mean a Johnny-Five IO Plugin for the ODROID-C1? If so, there are already a few IO Plugins for Linux boards which you could look at for ideas (which you already appear to be doing :)). There's also Gitter chats, one for [Johnny-Five](https://gitter.im/rwaldron/johnny-five) and a new one for [IO Plugins](https://gitter.im/rwaldron/io-plugins). The Gitter chats are ideal for ad hoc questions.

vallettea, 2015-02-22:
Thanks for the ressources.
I've moved to debian and the led works nice.
However I have trouble making a servo to move. Plugged on pin 10 (P9_42), i have a script : 

```
var five = require("johnny-five"),
    BeagleBone = require("beaglebone-io");

var board = new five.Board({
  io: new BeagleBone()
});

board.on("ready", function() {

  var servo = new five.Servo(10);
  servo.to(90, 500, 10);

});
```

which I launch in the BBB with `sudo node script.js`
Nothing happens and I find myself in a prompt:

```
debian@beaglebone:~$ sudo node script.js
1424646084730 Device(s) BeagleBone-IO
1424646084760 Connected BeagleBone-IO
1424646084790 Repl Initialized
>>
```

Any idea ? I doubt I have the good workflow.