julianduque/beaglebone-io #19: Beaglebone Black high CPU usage

mariobrizuela, 2015-08-19:
I just got my BBB to migrate a node project that I made for the Galileo Gen 2. Once I got my project running I noticed a very high cpu load, more than 2.0. Then I started to isolate the cause by stripping off the code until I get just the following code running;

var BeagleBone = require('beaglebone-io');
var board = new BeagleBone();
board.on('ready', function () {
  this.digitalWrite(13, this.HIGH);
});

Just the code above give me the following results;

top - 18:35:18 up 49 min,  2 users,  load average: 1.27, 1.44, 1.14

The node process was using around 98% of cpu. I'm running on Debian 3.8.13 and node v0.10.38. I haven't seen this problem reported by other beaglebone-io users, One interesting thing is that also galileo-io is showing the same behaviour in my Galileo Gen 2, but in this case the cpu load stays around 0.4, which is also pretty high.

fivdi, 2015-08-19:
The high CPU usage is probably because beaglebone-io uses the tick function to report pin status information as often/quickly as possible: https://github.com/julianduque/beaglebone-io/blob/master/lib/beaglebone.js#L107 

galileo-io reports pin status information every 10ms which is a lot slower and requires far less CPU time: https://github.com/rwaldron/galileo-io/blob/master/lib/galileo.js#L150

julianduque, 2015-08-20:
@fivdi we can change from `nexTick` to `setTimeout`, what do you think?

fivdi, 2015-08-20:
That sounds like a good idea to me.

mariobrizuela, 2015-08-20:
That would be great guys. In fact I made some tests wit setInterval in the read function, looks much better now, but still have high cpu times when I set the interval under 150ms, which is way to slow.
It would be nice to have a new version from you guys. Thanks a lot.

julianduque, 2015-08-20:
@mariobrizuela published 1.2.

mariobrizuela, 2015-08-21:
I still see %CPU of 97% and load average over 1.0. Maybe the setSamplingInterval can give better numbers? Thanks for your quick replies.

fivdi, 2015-08-21:
@mariobrizuela I'm currently no where near a BeagleBone so I can't test this myself. Are you absolutely sure the latest version of beaglebone-io is being used? Are you testing with the program from the first post above that only calls digitalWrite? Also, what is CPU usage for this process only and not the CPU usage for all processes as I think you're posting the CPU usage for all processes.

mariobrizuela, 2015-08-21:
Yes, I'm using the latest BeagleBone-io, the one with the read function based on setInterval.
No, I can't test it with the little code from my first post, because with the new BeagleBone-io node exits execution (without any fault) and goes to the system prompt. I made a new test with the following setup.

1. Set CPU frequency to 1000MHz to avoid any governor intervention.
2. Run my complete code already tested in Galileo Gen2.

This picture shows the result of TOP with BeagleBone-io based on the Tick function
[image]

This with setInterval function
[image]

There's almost no difference between the two versions.
Let me know if you want me to test something else.

julianduque, 2015-08-22:
Will find time this week to test with a proper beaglebone and see if we can improve this performance.. Thanks for the report

rwaldron, 2015-08-26:
FWIW, I had to make this same change to temporal, for the same reason.

fivdi, 2015-08-28:
Here are the results of a few simple tests.

The following, which doesn't really do very much, results in approximately 4% CPU usage:
```js
setInterval(function () {
}, 10);
```
The following, which reads the state of an input pin 100 times a second, results in approximately 40% CPU usage.
```js
var b = require("bonescript");

b.pinMode('P8_7', b.INPUT);

setInterval(function () {
  b.digitalRead('P8_7', function () {
  });
}, 10);
```
The second example uses BoneScript to read the state of an input pin the way BeagleBone-IO would do it. It appears to be the case that BoneScript isn't very fast at reading the state of an input pin and that the CPU is being used mostly by BoneScript.

fivdi, 2017-04-16:
I'm sorry it took so long to respond to this issue. The problems mentioned above are addressed by PR #32.

The original issue reported was that the following program resulted in 100% CPU usage.
```js
var BeagleBone = require('beaglebone-io');
var board = new BeagleBone();

board.on('ready', function () {
  this.digitalWrite(13, this.HIGH);
});
```
The equivalent of this program for PR #32 results in 0% CPU usage with Node.js v4.8.0 and the 4.4.54-ti-r93 Linux kernel.
```js
var BeagleBone = require('beaglebone-io');
var board = new BeagleBone();

board.on('ready', function () {
  this.pinMode('USR3', this.MODES.OUTPUT);
  this.digitalWrite('USR3', this.HIGH);
});

setTimeout(function() {}, 1e9);
```
Note that with PR #32 `setTimeout(function() {}, 1e9)` needs to be called in order to prevent the program from terminating after turning the USR3 LED on. Without the timeout the program correctly assumes that it has nothing to do after turning the LED on and terminates.

It was also reported that `digitalRead` wasn't particularly fast:

> Here are the results of a few simple tests.
>
> The following, which doesn't really do very much, results in approximately 4% CPU usage:
>
> ``` js
> setInterval(function () {
> }, 10);
> ```
>
> The following, [...]

At the time, these tests were performed with Node.js v0.10 and the sampling interval was 10ms.

Now Node.js v4.8.0 is being used and the sampling interval is 5ms rather than 10ms.

The following, which doesn't really do very much, results in approximately 6.6% CPU usage.
``` js
setInterval(function () {
}, 5);
```
The following BoneScript program which reads the state of an input pin 200 times a second results in approximately 31% CPU usage.
``` js
var b = require("bonescript");

b.pinMode('P8_7', b.INPUT);

setInterval(function () {
  b.digitalRead('P8_7', function () {
  });
}, 5);
```
The following BeagleBone-IO program with PR #32 which reads the state of an input pin 200 times a second results in approximately 14% CPU usage.
```js
var BeagleBone = require('beaglebone-io');
var board = new BeagleBone();

board.on('ready', function() {
  this.pinMode('P8_15', this.MODES.INPUT);
  this.digitalRead('P8_15', function () {
  });
});
```

julianduque, 2017-04-17:
v2.0.0 has been published, thanks to great work by @fivdi

closed 2017-04-17 by julianduque

mariobrizuela, 2017-04-18:
Congrats guys for the great job!! That´s a big step forward. Thank you all.

PR #20 (2015-08-20): change tick to setTimeout
Proposed PR to fix #19