julianduque/beaglebone-io #24: Change underlying library to OctalBoneScript

adityapatadia, 2015-12-28:
Hi,
I see this project is using BoneScript library. It's been abandoned since long and does not work on newer boards and OS. I request to change underlying library to https://github.com/theoctal/octalbonescript

julianduque, 2015-12-29:
Pull requests with tests are always welcome!

adityapatadia, 2015-12-29:
Hi, I would create a PR but don't have that much resources to move this library. I gave a suggestion as this library was using BoneScript which is not maintained anymore. If anyone integrated OBS and needs any help, I would be ready to do it.

Thanks,

mhelmetag, 2016-07-04:
@adityapatadia , bonescript seems to have fairly recent updates... whereas octalbonescript hasn't been updated since May (at the time of writing this). The rewrite is possible but I was wondering if you could start a conversation about the pros and cons...

fivdi, 2017-04-16:
@adityapatadia There are two things related to octalbonescript that would currently make it difficult for me to use it in BeagleBone-IO. The first is performance and the second is its dependency on the module i2c.

### Performance
Here is an octalbonescript program that toggles the state of an output pin as fast as it can and prints a message saying how often the state can be changed.
```js
var b = require("octalbonescript"),
  pin = 'P8_16',
  writesPerSecond,
  time,
  i;

b.pinModeSync(pin, b.OUTPUT);

time = process.hrtime();

for (i = 1; i <= 50000; i += 1) {
  b.digitalWriteSync(pin, i & 1);
}

time = process.hrtime(time);
writesPerSecond = Math.floor(i / (time[0] + time[1] / 1E9));

console.log(writesPerSecond + ' digitalWriteSync calls per second');
```
Here is typical output from the program.
```
3454 digitalWriteSync calls per second
```
Here is a BeagleBone-IO program that does the same thing.
```js
var five = require('johnny-five');
var BeagleBone = require('..');

var board = new five.Board({
  io: new BeagleBone()
});

board.on('ready', function() {
  var pin = board.io.normalize('GPIO46'),
    writesPerSecond,
    time,
    i;

  this.pinMode(pin, five.Pin.OUTPUT);

  time = process.hrtime();

  for (i = 1; i <= 250000; i += 1) {
    this.digitalWrite(pin, i & 1);
  }

  time = process.hrtime(time);
  writesPerSecond = Math.floor(i / (time[0] + time[1] / 1E9));

  console.log(writesPerSecond + ' digitalWrite calls per second');
});
```
Here is typical output from the program when run against #32 
```
86390 digitalWrite calls per second
```
The second program is 25 times faster that the first program.
### i2c
The module i2c goes through fairly long periods of low maintenance and drops support for versions of node that are still in fairly widespread usage in a non-graceful fashion. In addition, i2c doesn't provide async methods. It looks like it does provide async methods as it has methods with callbacks but under the covers these methods are actually blocking synchronous calls.

closed 2017-04-17 by julianduque

adityapatadia, 2017-04-18:
I am really surprised by speed difference because we use least code possible to get writes done. I will check on my own and get back to you. 

Anyway, I suggested because this project was using Bonescript. I can see that now you have moved to other libraries which are better maintained.

fivdi, 2017-04-18:
> I am really surprised by speed difference because we use least code possible to get writes done. I will check on my own and get back to you.

@adityapatadia I'm sure you'll be able to find things that can be improved.

In addition to using different libraries the architecture has also changed. Functionality common to multiple Linux boards is now located in [linux-io](https://github.com/fivdi/linux-io). BeagleBone-IO now only contains code specific to the BeagleBone Black.