julianduque/beaglebone-io #25: Install fails on BeagleBone Black - nodejs v0.12.13

mikewop, 2016-06-20:
- installed nodejs 0.12.13 on BeagleBone Black (directly from NodeSource https://github.com/nodesource/distributions)
- npm install beaglebone-io fails
```
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1078:12)
gyp ERR! System Linux 4.4.12-ti-r30
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/node-red/.node-red/node_modules/beaglebone-io/node_modules/bonescript/node_modules/ffi
gyp ERR! node -v v0.12.13
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
npm ERR! Linux 4.4.12-ti-r30
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "beaglebone-io"
npm ERR! node v0.12.13
npm ERR! npm  v2.15.0
npm ERR! code ELIFECYCLE

npm ERR! ffi@1.2.6 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the ffi@1.2.6 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the ffi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs ffi
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls ffi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/node-red/.node-red/npm-debug.log
```
 seems like the culprit is the ffi package.
However 
```npm install ffi```
 worked without error, which installs ffi 2.0.0. Seems like beaglebone-io tries to use ffi 1.2.6

hadrien-toma, 2016-08-17:
Same error for me, and it also occurs with Node.js 4.x, 5.x and 6.x.

hethcox, 2016-10-09:
Is there a workaround for this?

justindstewart, 2016-11-06:
Still having issues even after trying @abythell fix.

abythell, 2016-11-08:
@justindstewart - Care to describe those issues?

justindstewart, 2016-11-08:
Absolutely. Fresh install of Debian Image 2016-05-13. I run apt-get update/upgrade. Upgrade Node to 7. I then try upgrading Bonescript and i2c-bus as you instructed and I still get the ffi failure like above. I can post a debug log once I get home.

troywweber7, 2017-01-26:
Any solutions to this? This is happening on the BeagleBone Green Wireless as well. Issue #29

troywweber7, 2017-01-27:
I don't know much about troubleshooting this stuff because I'm kind of a newb, but since it is using the wrong `ffi` version, if I were to fork `beaglebone-io` and change the package to `ffi@2.0.0` and then try installing my fork, is that a viable work around? Could that work? Or are changes between `ffi@2.0.0` and `ffi@1.2.6` potentially too breaking for it to be that simple?

abythell, 2017-01-27:
@troywweber7 - beaglebone-io doesn't depend directly on ffi - it's a dependency of bonescript.  Applying #25 updates the version of bonescript, which depends on a newer version of ffi.  It uses bonescript@0.5.1,  i2c@0.2.1 and ffi@2.0.0.

I have just re-confirmed that #25 builds successfully on node v0.12.17, which is the version installed out-of-the-box in the Debian 2016-11-06 image. @stewartjustind 's issues may be due to node v7 but can't say with seeing the output of the build.

Here's how I'm building it:
```
node -v  # v0.12.17
npm -v  # 2.15.1
cat /proc/version
  #Linux version 4.4.30-ti-r64 (root@a2-imx6q-wandboard-2gb) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1     
  #SMP Fri Nov 4 21:23:33 UTC 2016
git clone https://github.com/julianduque/beaglebone-io.git
cd beaglebone-io
git remote add abythell git:///github.com/abythell/beaglebone-io.git
git fetch abythell
git cherry-pick 4d87a747eb3930d82ff29a01f7704106e85a6ee2 --no-commit
npm install
```

troywweber7, 2017-01-27:
@abythell Thanks for responding.

> Applying #25 updates the version of bonescript, which depends on a newer version of ffi

What is meant by "Applying #25" in this sentence? #25 references this issue, and it's not clear to me what you mean by "applying #25".

> I have just re-confirmed that #25 builds successfully on node v0.12.17

Same question as above. Do you mean you reconfirmed that `npm i -g ffi` installs fine on node v0.12.17?

abythell, 2017-01-27:
Sorry, that should have been #27, as in, "apply the patch described in pull request #27, which should fix issue #25".

In other words, if you follow the step-by-step instructions posted in my previous comment, or simply edit package.json and change the dependencies to bonescript@0.5.0 and i2c-bus@1.1.2, you will hopefully be able to build the beaglebone-io package.

troywweber7, 2017-01-27:
Just curious: why `bonescript@0.5.0` and not `bonescript@0.5.1`?

troywweber7, 2017-01-27:
> ```
> node -v  # v0.12.17
> npm -v  # 2.15.1
> cat /proc/version
>   #Linux version 4.4.30-ti-r64 (root@a2-imx6q-wandboard-2gb) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1     
>   #SMP Fri Nov 4 21:23:33 UTC 2016
> ```

Just to be clear, we should be in this directory, right? Or essentially wherever your global node-modules are stored?
```
cd /usr/local/lib
```

> ```
> git clone https://github.com/julianduque/beaglebone-io.git
> cd beaglebone-io
> git remote add abythell git:///github.com/abythell/beaglebone-io.git
> git fetch abythell
> git cherry-pick 4d87a747eb3930d82ff29a01f7704106e85a6ee2 --no-commit
> npm install
> ```

abythell, 2017-01-27:
0.5.0 was the latest version back in October when this was first proposed.  I suppose to be unambiguous the bonescript version in package.json should read `^0.5.0`.

You can build in any directory you want.  I'd probably use $HOME, then `npm link` to use the newly built package in other projects.

troywweber7, 2017-01-27:
@abythell thanks for the suggestion to use `npm link`. I'm pretty new to this stuff, so that gives me something to read up on. However, I was able to install it by going in to the directory, git-cloning, and changing the `package.json` to

```
  "dependencies": {
    "bonescript": "^0.5.0",
    "i2c-bus": "^1.1.2"
  }
```

It `npm install`ed just fine after that. Here is my system deets

```
root@beaglebone:~# uname -a
Linux beaglebone 4.4.30-ti-r64 #1 SMP Fri Nov 4 21:23:33 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# cat /etc/debian_version 
8.7
root@beaglebone:~# node -v
v0.12.18
root@beaglebone:~# npm -v
2.15.11
root@beaglebone:~# npm list -g beaglebone-io bonescript i2c-bus
/usr/local/lib
├─┬ beaglebone-io@1.3.0 
│ ├── bonescript@0.5.1 
│ └── i2c-bus@1.2.0 
├── bonescript@0.5.1 
└─┬ wificonfig@0.1.5
  └── bonescript@0.5.2 
```

This is a good start for me, but I will likely try getting this working under node 4/6/7. My projects tend to be node 6 and up. If I have success with that, I'll try to post details here.

abythell, 2017-01-30:
Starting from a fresh 2016-11-05 image I have confirmed that increasing the version of bonescript as proposed in PR #27 will resolve this issue on node v0.12.13, v6.9.4, and v7.4.0 (tested but not shown).

Test method for v6.9.4:

```
apt-get update
apt-get upgrade -y
curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
sudo apt-get install -y nodejs
node -v # v6.9.4
npm -v # 3.10.10
git clone https://github.com/julianduque/beaglebone-io.git
cd beaglebone-io
git remote add abythell git:///github.com/abythell/beaglebone-io.git
git fetch abythell
git cherry-pick 4d87a747eb3930d82ff29a01f7704106e85a6ee2 --no-commit
npm install
```

troywweber7, 2017-01-31:
@abythell Good to know, thank you!

fivdi, 2017-04-16:
PR #32 addresses this issue and several other issues. If PR #32 is merged, BeagleBone-IO will support Node.js v0.10, v0.12, v4, v5, v6 and v7.

julianduque, 2017-04-17:
v2.0.0 has been published, thanks to great work by @fivdi

closed 2017-04-17 by julianduque