julianduque/beaglebone-io #26: Error: Module did not self-register when running basic LED code on Beaglebone Black (Node v6.2.2)

kongxingagit, 2016-07-27:
Hi,

New to Node, Javascript and Beaglebone Black in general.

I've been trying to run very basic LED blinking code as follows. What I want to do is blink an LED attached to my Beaglebone Black. I am running this with direct power to the board which is also connected to Ethernet via Ethernet cable to my router (ie not internet through USB). The code is being run in the Cloud9 IDE that comes with the Beaglebone.

```javascript
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
    io: new BeagleBone()
});

board.on('ready', function () {
    // connect a Led on pin P9_14 of the beaglebone
    var led = new five.Led('P9_14');
    led.blink(1000);
    // this allows access to the led while the program is running
    this.repl.inject({ led: led });
});
``` 
I believe this should blink an LED that is attached to ground and pin P9_14 (attached via breadboard and suitable resistor).

Instead this is throwing the following error which I reproduce.

```
/usr/local/lib/node_modules/beaglebone-io/node_modules/bonescript/node_modules/ffi/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: Module did not self-register.
    at Error (native)
    at Object.Module._extensions..node (module.js:568:18)
    at Module.load (module.js:458:32)
    at tryModuleLoad (module.js:417:12)
    at Function.Module._load (module.js:409:3)
    at Module.require (module.js:468:17)
    at require (internal/module.js:20:19)
    at bindings (/usr/local/lib/node_modules/beaglebone-io/node_modules/bonescript/node_modules/ffi/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/usr/local/lib/node_modules/beaglebone-io/node_modules/bonescript/node_modules/ffi/node_modules/ref/lib/ref.js:5:47)
    at Module._compile (module.js:541:32)

[Process stopped]
```

I googled the Error: Module did not self register. One hit suggested running 
```
npm rebuild
```
in terminal, which I did for the beaglebone but the error persisted.

The rest of the hits were about certain libraries not supporting newer node versions.
To follow that, I ran 

```
node --version

```

This returned v6.2.2 which I guess is a pretty new but probably not current version of node.

Is there anything I can do to further diagnose this issue? Am I on the right track?  Are there possible workarounds where I have a local copy of an older version of node that would play well with beaglebone-io? I am quite green at this, and could be doing multiple things wrong. I don't quite have the ability to determine if it is a beaglebone, hardware, node, johnny-five or other issue at my level.

fivdi, 2017-04-16:
I'm sorry for taking so long to respond to this issue. PR #32 addresses this issue and several others. If PR #32 is merged, BeagleBone-IO will support Node.js v0.10, v0.12, v4, v5, v6 and v7.

julianduque, 2017-04-17:
v2.0.0 has been published, thanks to great work by @fivdi

closed 2017-04-17 by julianduque