julianduque/beaglebone-io #29: SeeedStudio BeagleBone Green Wireless install fails

troywweber7, 2017-01-24:
Installation of BeagleBone-IO (`npm install --save beaglebone-io`) fails. Here are some details:

- Using SeeedStudio BeagleBone Green Wireless
- `uname -a` produces `Linux [hostname redacted] 4.4.30-ti-r64 #1 SMP Fri Nov 4 21:23:33 UTC 2016 armv7l GNU/Linux`
- `lsb_release -a` produces

```
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.7 (jessie)
Release:	8.7
Codename:	jessie
```

- `node -v` = `v7.4.0`
- `npm -v` = `4.0.5`

Does anyone know about this issue already? Is there an obvious issue based on the given information?

If not, is there more information I can provide to help track down the issue? 

Here is the npm-debug.log, below:

```
0 info it worked if it ends with ok
1 verbose cli [ '/home/deploy/.nvm/versions/node/v7.4.0/bin/node',
1 verbose cli   '/home/deploy/.nvm/versions/node/v7.4.0/bin/npm',
1 verbose cli   'install',
1 verbose cli   '--save',
1 verbose cli   'beaglebone-io' ]
2 info using npm@4.0.5
3 info using node@v7.4.0
4 silly loadCurrentTree Starting
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 silly fetchPackageMetaData beaglebone-io
8 silly fetchNamedPackageData beaglebone-io
9 silly mapToRegistry name beaglebone-io
10 silly mapToRegistry using default registry
11 silly mapToRegistry registry https://registry.npmjs.org/
12 silly mapToRegistry data Result {
12 silly mapToRegistry   raw: 'beaglebone-io',
12 silly mapToRegistry   scope: null,
12 silly mapToRegistry   escapedName: 'beaglebone-io',
12 silly mapToRegistry   name: 'beaglebone-io',
12 silly mapToRegistry   rawSpec: '',
12 silly mapToRegistry   spec: 'latest',
12 silly mapToRegistry   type: 'tag' }
13 silly mapToRegistry uri https://registry.npmjs.org/beaglebone-io
14 verbose request uri https://registry.npmjs.org/beaglebone-io
15 verbose request no auth needed
16 info attempt registry request try #1 at 7:12:19 PM
17 verbose request id d8bd211899bfe361
18 verbose etag W/"55e5e7e1-58c4"
19 verbose lastModified Tue, 01 Sep 2015 18:01:05 GMT
20 http request GET https://registry.npmjs.org/beaglebone-io
21 http 304 https://registry.npmjs.org/beaglebone-io
22 verbose headers { date: 'Tue, 24 Jan 2017 19:12:19 GMT',
22 verbose headers   via: '1.1 varnish',
22 verbose headers   'cache-control': 'max-age=300',
22 verbose headers   etag: 'W/"55e5e7e1-58c4"',
22 verbose headers   age: '0',
22 verbose headers   connection: 'keep-alive',
22 verbose headers   'x-served-by': 'cache-lax8623-LAX',
22 verbose headers   'x-cache': 'MISS',
22 verbose headers   'x-cache-hits': '0',
22 verbose headers   'x-timer': 'S1485285139.908781,VS0,VE31',
22 verbose headers   vary: 'Accept-Encoding' }
23 silly get cb [ 304,
23 silly get   { date: 'Tue, 24 Jan 2017 19:12:19 GMT',
[... 3109 lines cut ...]
2023 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ref-struct
2024 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ref-struct
2025 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2026 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2027 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2028 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2029 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2030 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2031 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2032 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/ffi
2033 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2034 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2035 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2036 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2037 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2038 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2039 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2040 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/bonescript
2041 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2042 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2043 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2044 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2045 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2046 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_modules/beaglebone-io
2047 silly vacuum-fs purging /home/deploy/Code/JsTesting/node_

troywweber7, 2017-01-24:
Also, please note: the readme says "If a directory matching the pattern /sys/devices/bone_capemgr.* exists, the required support is available." What if that directory doesn't exist? Is there a way to tell if you have support otherwise?

troywweber7, 2017-01-24:
Okay, I use nvm, so I quickly rolled back to node version 6, and I can now install beaglebone-io. Any ideas on when it will be node version 7 compatible?

troywweber7, 2017-01-26:
Likely a duplicate issue to #25. Really need a workaround to this or something.

fivdi, 2017-04-16:
@troywweber7 I apologize for taking so long to respond to this issue. 

>the readme says "If a directory matching the pattern /sys/devices/bone_capemgr.* exists, the required support is available." What if that directory doesn't exist? Is there a way to tell if you have support otherwise?

Unfortunately the readme isn't up to date as this comment applies to an older version of the Linux kernel and is not valid for the 4.4 kernel which is the kernel that you are using. The current version of BeagleBone-IO doesn't support the 4.4 kernel.

If you are still interested in using beagleBone-IO PR #32 adds support for the 4.4 kernel and should address the issues that you are experiencing. It supports Node.js v0.10, v0.12, v4, v5, v6 and v7. If you would like to give it a try the PR can be installed with

```npm install fivdi/beaglebone-io#rewrite```

The readme is [here](https://github.com/fivdi/beaglebone-io/tree/rewrite).

That being said, BeagleBone-IO was implemented for the BeagleBone Black and I'm not sure to what extent it will support the BeagleBone Green Wireless. However, if you want to give it a try I will attempt to help with any issue related to the topic.

fivdi, 2017-04-20:
BeagleBone-IO v2 has been released and supports Node.js v0.10, v0.12, v4, v5, v6 and v7. There is a fairly good chance that it will function on a BeagleBone Green.

fivdi, 2018-01-10:
BeagleBone-IO installs successfully on a BeagleBone Green Wireless. See console output below.

That being said, the pin definitions for the BeagleBone Black will be used at runtime. To improve support for the BeagleBone Green Wireless BeagleBone-IO would have to be extended to include pin definitions for the BeagleBone Green Wireless.

See #43

```
debian@beaglebone:~/beaglebone-io$ cat /proc/device-tree/model 
TI AM335x BeagleBone Green Wireless
```
```
debian@beaglebone:~/beaglebone-io$ uname -a
Linux beaglebone 4.4.91-ti-r133 #1 SMP Tue Oct 10 05:18:08 UTC 2017 armv7l GNU/Linux
```
```
debian@beaglebone:~/beaglebone-io$ node -v
v6.11.4
```
```
debian@beaglebone:~/beaglebone-io$ npm install beaglebone-io

> epoll@1.0.2 install /home/debian/beaglebone-io/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/debian/beaglebone-io/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/debian/beaglebone-io/node_modules/epoll/build'

> i2c-bus@1.2.5 install /home/debian/beaglebone-io/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/debian/beaglebone-io/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/debian/beaglebone-io/node_modules/i2c-bus/build'
/home/debian/beaglebone-io
└─┬ beaglebone-io@2.3.0 
  ├─┬ glob@7.1.2 
  │ ├── fs.realpath@1.0.0 
  │ ├─┬ inflight@1.0.6 
  │ │ └── wrappy@1.0.2 
  │ ├── inherits@2.0.3 
  │ ├─┬ minimatch@3.0.4 
  │ │ └─┬ brace-expansion@1.1.8 
  │ │   ├── balanced-match@1.0.0 
  │ │   └── concat-map@0.0.1 
  │ ├── once@1.4.0 
  │ └── path-is-absolute@1.0.1 
  └─┬ linux-io@0.8.1 
    ├─┬ board-io@3.0.5 
    │ └── check-types@1.4.0 
    ├─┬ i2c-bus@1.2.5 
    │ ├── bindings@1.3.0 
    │ └── nan@2.8.0 
    ├── mutexify@1.2.0 
    └─┬ onoff@1.1.9 
      └── epoll@1.0.2 

npm WARN enoent ENOENT: no such file or directory, open '/home/debian/beaglebone-io/package.json'
npm WARN beaglebone-io No description
npm WARN beaglebone-io No repository field.
npm WARN beaglebone-io No README data
npm WARN beaglebone-io No license field.
debian@beaglebone:~/beaglebone-io$ 
```

closed 2018-01-10 by fivdi

fivdi, 2018-01-22:
BeagleBone-IO now has first class support for the BeagleBone Green Wireless.