julianduque/beaglebone-io #31: Should BeagleBone-IO be rewritten?

fivdi, 2017-02-15:
@julianduque and anyone else, would it be a good idea to rewrite BeagleBone-IO?

Currently BeagleBone-IO is a little out of date and could probably be brought up to date by updating a few dependencies (#27.)

However, what I'm thinking of would go a little further and has two motivations.

The first motivation is to resolve as many of the current issues as possible:

  * Installation issues (#25, #26, ...)
  * Additional pin support (#9)
  * Performance issues (#19.) Implicitly, this means parting from BoneScript which according to the repo is [totally unoptimized](https://github.com/jadonk/bonescript#note-on-performance). A few digital IO tests indicate that OctalBoneScript isn't more performant than BoneScript (#24.)

The second motivation is that I recently implemented [linux-io](https://github.com/fivdi/linux-io) and want to see how suitable it is for usage as an extensible Linux IO Plugin for Johnny-Five.

Note that the title of this issue is "Should BeagleBone-IO be rewritten?" and not "Should BeagleBone-IO be refactored?". With refactoring the external behavior doesn't change, with rewriting it typically does. An example of what might be worth changing here is the Arduino pin numbering. For example, on a BeagleBone Black `P8_7` may be better that `0` as a pin ID.

An additional point that should be mentioned is that what I'd like to do here would only be supported on the 4.4 kernel. Older 3.8 and 4.1 kernels would not be supported. One of the advantages of BoneScript is that it bends over backwards to support as many kernels as possible.

So, is this a good idea?

MrYsLab, 2017-02-15:
@fivdi 
I think a rewrite at this time is totally appropriate.

julianduque, 2017-02-16:
@fivdi I'm 100% agree with this suggestion and we can publish it as a 2.x branch with support only for 4.4 kernel

fivdi, 2017-03-28:
Another issue that should be resolved with the rewrite is

 * Add own-instance properties: `aref` and `vref` (#21)

fivdi, 2017-03-28:
@julianduque I'm going to start working on this in the not too distant future, ok?

julianduque, 2017-03-29:
@fivdi sounds good to me, let me know if there is anything I can help with

julianduque, 2017-04-17:
v2.0.0 has been published, thanks to great work by @fivdi

closed 2017-04-17 by julianduque