julianduque/beaglebone-io #33: Pin numbers for Johnny Five compatibility

mariobrizuela, 2017-04-20:
As beaglebone-io v2 has dropped support for Arduino pin numbers I wonder how Johnny Five would deal with it. As we all know, Johnny Five doesn´t use the PX_XX or GPIOXX nomenclature, just plain numbers. According to what I found, V2 put all Arduino pins in P8 from 7 to 19. For analog inputs nothing changed.

[image]

Would be nice if you can confirm this information and maybe put it in the readme.
Thank you guys.

fivdi, 2017-04-20:
Nice to see someone using v2 :)

Johnny-Five supports the usage of the Pin IDs shown in the [Supported Pins table](https://github.com/julianduque/beaglebone-io#supported-pins). There are a few programs in the [examples](https://github.com/julianduque/beaglebone-io/tree/master/examples) directory that show how to do it, for example, to pulse an LED in P8_13:
```js
var five = require('johnny-five');
var BeagleBone = require('..');

var board = new five.Board({
  io: new BeagleBone()
});

board.on('ready', function() {
  var led = new five.Led('P8_13');

  led.pulse(1000);
});
```

mariobrizuela, 2017-04-20:
@fivdi so sorry to waste your time. I think is time for me to stop looking at this screen and go for the glass of wine waiting at home. Thank you again.

fivdi, 2017-04-20:
Enjoy the wine.

I think the readme could be modified to improve things. Currently there are no examples in the usage section using  PX_XX or GPIOXX.

fivdi, 2017-04-21:
PR #34 addresses this issue.

julianduque, 2017-04-23:
With documentation improvements we can close this issue, thank you @fivdi and @mariobrizuela !

closed 2017-04-23 by julianduque