julianduque/beaglebone-io #44: U-Boot, v4.4.x/v4.9.x/v4.14.x: and v4.15.x+ support

RobertCNelson, 2018-01-10:
So recently beagleboard.org dumped the idea of using Kernel Overlays and have replaced it with U-Boot Overlays. (just too many bugs to deal with)

Things to watch out for:

U-Boot signals to the Kernel that it has control thru "/proc/cmdline"

So if, "bone_capemgr.uboot_capemgr_enabled=1" is set in /proc/cmdline do not utilize slots file:

So if that signal thru /proc/cmdline, some important things for v4.4.x/v4.9.x/v4.14.x

v4.4.x/v4.9.x/v4.14.x:

If slots exists (/sys/devices/platform/bone_capemgr/slots), adding or removing overlays when U-Boot set things up will cause a kernel lockup...

If slots file is missing (it was removed to fix accidental hits ^ and is removed by bone_capemgr.uboot_capemgr_enabled=1 ) the /sys/devices/platform/bone_capemgr/ directory will still exist

v4.15.x:

In v4.15.x+ the /sys/devices/platform/bone_capemgr/ is no more.. (U-Boot Overlays only option)

Assumptions with U-Boot Overlays:

ADC  = always enabled
cape-universal enabled by default: thus every IO available (thru config-pin).

bone_capemgr directory removed

available gpio's enabled can be search via:

"/sys/devices/platform/ocp/ocp:P*_*_pinmux"

BeagleBoard supports:

"gpio gpio_pu gpio_pd" options for "/sys/devices/platform/ocp/ocp:P*_*_pinmux/state"

PocketBeagle supports

"gpio gpio_pu gpio_pd gpio_input" options for "/sys/devices/platform/ocp/ocp:P*_*_pinmux/state"

P9_19:P9_20: now carved out, and no longer 100% tied to the cape-eeprom, it's only needed in u-boot, thus can be used as can/gpio/etc in kernel land.

PS: i'm working on integrating this into our "node-red" deb package. ;)

Regards,

fivdi, 2018-01-10:
@RobertCNelson thanks for all the tips, they'll come in very useful. BeagleBone-IO only uses slots in one place now, to enable ADC if it isn't already enabled. If it's always enabled then that last usage of slots can be removed. If `disable_uboot_overlay_adc=1` is in uEnv.txt it was explicitly disabled so that will be ok too.

> PS: i'm working on integrating this into our "node-red" deb package. ;)

That's great news :smile: Will you also be integrating [Johnny-Five](https://github.com/rwaldron/johnny-five)? BeagleBone-IO is a plugin for Johnny-Five and is infinitely more useful with Johnny-Five. I'll have to admit that I'm not very familiar with node-red or the node-red deb package. Is BeagleBone-IO being added to the node-red deb package to increase the functionality of node-red? Or is BeagleBone-IO being added to the node-red deb package so that it can be installed and used independently of node-red?

RobertCNelson, 2018-01-10:
@fivdi correct, it's already linked up in buster as of today, as part of the node-red package. (With Johnny-Five)

https://github.com/rcn-ee/repos/blob/master/debian-10-bb-node-red-installer/version.sh#L22-L26

https://prnt.sc/hyqj7o

While it is part of the node-red package, it's in it's own directory, so i'm going to seperate our current node-red into 3 packages, beaglebone-io, johnny-five, and node-red (node-red/node-red-contrib-gpio)

```
debian@beaglebone:/usr/local/lib/node_modules$ ls -lha
total 0
drwxr-sr-x 1 root   staff   210 Jan 10 21:39 .
drwxrwsr-x 1 root   staff    60 Jan  7 16:45 ..
drwxr-sr-x 1 root   staff  1.8K Dec  4 20:19 async
drwxr-sr-x 1 debian debian  162 Jan 10 21:32 beaglebone-io
drwxr-sr-x 1 debian debian  230 Jan 10 02:32 bonescript
drwxr-sr-x 1 debian debian  118 Jan 10 20:53 johnny-five
drwxr-sr-x 1 debian debian  344 Jan 10 20:02 node-red
drwxr-sr-x 1 debian debian  208 Jan 10 20:54 node-red-contrib-gpio
drwxr-sr-x 1 debian debian  150 Jan 10 20:03 node-red-node-mongodb
drwxr-sr-x 1 root   staff   226 Dec  4 20:21 sensortag
drwxr-sr-x 1 debian debian  118 Jan 10 02:35 winston
```

and then backport that to stretch.

Regards,

fivdi, 2018-01-10:
Nice, very nice indeed :smile:

fivdi, 2018-01-14:
PR #49 removes all usage of the slots file from BeagleBone-IO.

fivdi, 2018-01-22:
@RobertCNelson 

BeagleBone-IO has been updated and no longer uses `slots` or `/sys/devices/platform/bone_capemgr/`. 

This means that it should function correctly on v4.4.x/v4.9.x/v4.14.x/v4.15.x. However, I'll have to admit that I only tested on v4.4 and v4.14. 

BeagleBone-IO doesn't require root privileges on v4.14.x.

P9_19:P9_20 are used by BeagleBone-IO for I2C. This has always been the case although is sounds like this may not have been a good practice.

Two follow up issues based on the information you provided above have been created, #59 and #60.

Please feel free to create issues similar to this issue in the future, it would be very helpful.

I'll leave this issue open for the moment and close it after beaglebone-io@3.0.0 has been published on npm.

RobertCNelson, 2018-01-22:
@fivdi awesome!

ps, i snuck the git version in on Saturday night before the weekly build (Sunday morning), so the this weekly build has up to git:

https://github.com/julianduque/beaglebone-io/commit/8d4e9f461243e80741c7fee06f7c84041409f0bc

https://rcn-ee.net/rootfs/bb.org/testing/2018-01-21/

After the two pull requests this morning, i triggered another build, i'll have updated *.deb packages in about another 1/2 hour.  So Stretch & Buster are looking great! ;)

Regards,

fivdi, 2018-01-22:
@RobertCNelson beaglebone-io@3.0.0 has been published on npm.

> After the two pull requests this morning, i triggered another build, i'll have updated *.deb packages in about another 1/2 hour.

This means that beaglebone-io@3.0.0 was available in the deb package before it was published on npm.

closed 2018-01-22 by fivdi