julianduque/beaglebone-io #53: Add BeagleBone Blue pin mapping

roblamoreaux, 2018-01-19:
It would be really nice if the BeagleBone Blue pin names could be used like the PocketBeagle Pin names. I can't use the equivalent BBBlack or GPIOx_x pins and get it to work.

fivdi, 2018-01-20:
@roblamoreaux Which features of the BeagleBone Blue would you like to use with BeagleBone-IO / Johnny-Five?

I don't have a BeagleBone Blue to test with so could you post the output of the command `/sys/devices/platform/ocp` please?

roblamoreaux, 2018-01-20:
I'd love to be able to use the servos, the motor control and the encoder input. eventually I'd like to get Node-Red working so I can use it to help my kids learn to program the various robot bits I have around the house, but I am finding the support for the Blue disappointing.

Anyway here is the directory listing:

debian@beaglebone:~$ ls -l /sys/devices/platform/ocp
total 0
drwxr-xr-x 3 root root    0 Jan  1  2000 40300000.ocmcram
drwxrwxr-x 5 root gpio    0 Jan  1  2000 44e07000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 44e09000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 44e0b000.i2c
drwxr-xr-x 4 root root    0 Jan  1  2000 44e0d000.tscadc
drwxr-xr-x 5 root root    0 Jan  1  2000 44e35000.wdt
drwxr-xr-x 4 root root    0 Jan  1  2000 44e3e000.rtc
drwxr-xr-x 9 root root    0 Jan  1  2000 47400000.usb
drwxr-xr-x 4 root root    0 Jan  1  2000 47810000.mmc
drwxr-xr-x 4 root root    0 Jan  1  2000 48022000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 48024000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 4802a000.i2c
drwxr-xr-x 3 root root    0 Jan  1  2000 48042000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48044000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48046000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48048000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 4804a000.timer
drwxrwxr-x 5 root gpio    0 Jan  1  2000 4804c000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 48060000.mmc
drwxr-xr-x 4 root root    0 Jan  1  2000 480c8000.mailbox
drwxr-xr-x 3 root root    0 Jan  1  2000 480ca000.spinlock
drwxr-xr-x 4 root root    0 Jan  1  2000 4819c000.i2c
drwxr-xr-x 3 root root    0 Jan  1  2000 481a0000.spi
drwxr-xr-x 4 root root    0 Jan  1  2000 481a6000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 481a8000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 481aa000.serial
drwxrwxr-x 5 root gpio    0 Jan  1  2000 481ac000.gpio
drwxrwxr-x 5 root gpio    0 Jan  1  2000 481ae000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 481d0000.can
drwxr-xr-x 4 root root    0 Jan  1  2000 481d8000.mmc
drwxr-xr-x 5 root root    0 Jan  1  2000 48300000.epwmss
drwxr-xr-x 5 root root    0 Jan  1  2000 48302000.epwmss
drwxr-xr-x 5 root root    0 Jan  1  2000 48304000.epwmss
drwxr-xr-x 3 root root    0 Jan  1  2000 48310000.rng
drwxr-xr-x 4 root root    0 Jan  1  2000 49000000.edma
drwxr-xr-x 3 root root    0 Jan  1  2000 49800000.tptc
drwxr-xr-x 3 root root    0 Jan  1  2000 49900000.tptc
drwxr-xr-x 3 root root    0 Jan  1  2000 49a00000.tptc
drwxr-xr-x 4 root root    0 Jan  1  2000 4a326000.pruss-soc-bus
drwxr-xr-x 3 root root    0 Jan  1  2000 4c000000.emif
drwxr-xr-x 3 root root    0 Jan  1  2000 53100000.sham
drwxr-xr-x 3 root root    0 Jan  1  2000 53500000.aes
-rw-r--r-- 1 root root 4096 Jan 20 23:49 driver_override
-r--r--r-- 1 root root 4096 Jan 20 23:49 modalias
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:C18_pinmux
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:D13_pinmux
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:H17_pinmux
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:H18_pinmux
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:helper
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:J15_pinmux
drwxr-xr-x 6 root root    0 Jan  1  2000 ocp:l4_wkup@44c00000
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P8_15_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_11_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_21_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_22_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_23_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_24_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_26_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_28_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_29_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_30_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_31_pinmux
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:U16_pinmux
lrwxrwxrwx 1 root root    0 Jan 20 23:49 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Jan 20 23:49 power
lrwxrwxrwx 1 root root    0 Jan  1  2000 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Jan  1  2000 uevent

techyjoe, 2018-04-21:
Any new developments on beagle bone blue support?

fivdi, 2018-04-21:
Sorry, nothing new to report here. I don't think support for the BeagleBone Blue will be added any time in the near future.

TheGreatJK, 2018-07-11:
Still no plans for BeagleBone Blue support? I just got one, and am really struggling to find a good way to make use of it's features from JavaScript.

RobertCNelson, 2018-07-11:
So out of the gate with the Blue, things got pushed to make it BBB + Blue extra.. Looking back, we should have just did a clean break like the PocketBeagle..

Here are my notes one possible way...

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/pinmux-generator/BeagleBone_Blue.dts

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/pinmux-generator/BeagleBone_Blue_config-pin.txt

Regards,

TheGreatJK, 2018-07-11:
Thanks for the links!

This would be a good time to admit I'm a _mechanical_ engineering student with a side hobby, quite a beginner in this field. How exactly would I implement such a library? 

I agree that it would be a pity to entirely circumvent libroboticape, so I was hoping to use something more like this:

https://github.com/jadonk/node-roboticscape

Unfortunately, they don't support the servo or gpio outputs yet, and those are exactly what I want to use. I built a web remote control application (with joystick support) on my old BeagleBone Black (entirely within JavaScript) and was hoping to switch everything to this board for the convenient on-board wifi and hardware interfaces.