julianduque/beaglebone-io #62: `analogRead` callback fires infinitely rather than once

abcd-ca, 2018-02-22:
I init my board like this:
`const BeagleBone = require('beaglebone-io')`
...
`this.board = new BeagleBone()`

on ready, I do this,

```
    this.board.pinMode(PIN_BATTERY, this.board.MODES.ANALOG)

    this.board.analogRead('A3', (value) => {
      console.log('*** analog value: ', value)
    })
```

**Expected result:**

I see, "*** analog value: 123"

**Actual result:**

I see, 

```
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
*** analog value: 123
(continues infinitely)
```

The numbers vary slightly as one might expect. Is this expected behaviour or should I only get one callback fn call per `analogRead` call?

abcd-ca, 2018-02-22:
update, if I try removing the try/catch block in the analogRead function source I discovered that this line,

`callback(null, pinData.custom.analogInput.rawValue() >> 2);`

is received by my callback handler with only one arguments (measured with `arguments.length`). I don't even know how that's possible. Thoughts?

fivdi, 2018-02-22:
> analogRead callback fires infinitely rather than once
> Is this expected behaviour or should I only get one callback fn call per analogRead call?

This is the expected behavior. Calling `analogRead` initiates a new continuous data reading process for a pin not a single read. BeagleBone-IO is an IO Plugin for Johnny-Five and this is how the `analogRead` method is expected to function. Further details can be seen at https://github.com/rwaldron/io-plugins#analogreadpin-handler.

> the examples show a callback with one argument for the value. However, the source code shows this:
>
> BeagleBone.prototype._analogRead = function(pinData, callback) {
>  setImmediate(function () {
>    try {
>      // Raw value is 0 to 4095 but J5 expects 0 to 1023.
>      callback(null, pinData.custom.analogInput.rawValue() >> 2);
>    } catch (err) {
>      callback(err);
>    }
>  });
>};

There is a misunderstanding here. As can be seen [here](https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/beaglebone.js#L31), BeagleBone-IO is a class that extends LinuxIO.

Your code calls `this.board.analogRead` which is implemented in LinuxIO by [LinuxIO.prototype.analogRead](https://github.com/fivdi/linux-io/blob/v0.9.2/lib/linux-io.js#L176). Your code is not calling `BeagleBone.prototype._analogRead` which is a private method of BeagleBoneIO implemented by [BeagleBone.prototype._analogRead](https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/beaglebone.js#L60).

abcd-ca, 2018-02-22:
Thank you, that clears things up. With such a bombardment of callback calls I wonder if you have a sense of the performance implications? What I might do is just update a numeric variable on the callback and then occasionally read from that variable. Still, I wonder how expensive such frequent callback calls will be on power/cpu on beaglebone?

I think while it's expected behaviour, it's not a typical expectation of that callback syntax. I'll create a PR with a little note in the home page docs to help people like me in the future.

abcd-ca, 2018-02-22:
PR is here: https://github.com/julianduque/beaglebone-io/pull/63

fivdi, 2018-02-24:
> With such a bombardment of callback calls I wonder if you have a sense of the performance implications?
> Still, I wonder how expensive such frequent callback calls will be on power/cpu on beaglebone?

If you need this information I'd suggest performing the required tests to see if BeagleBone-IO meets your requirements.

Thanks for the PR :)

I'm going to go ahead and close this issue as the questions from the OP have been answered.

closed 2018-02-24 by fivdi