julianduque/beaglebone-io #64: i2cWrite error on Debian 9.2:  Cannot read property 'i2cWriteSync' of undefined

totem415, 2018-03-16:
Would love for this to be working, but sadly, no.  

I'm on a Beaglebone Black, Debian 9.2, Node JS v6.11.4, running the latest beaglebone-io / johnny five combo.  I'm very confident that the i2c port has been activated in Debian, so that's not the problem.

Here's my code to initialize:
//--------------------------------------
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');
var boardReady = false;

var board = new five.Board({
  io: new BeagleBone()
});

 
board.on('ready', function() {
  console.log("i2cMgr:  johnny five board.on('ready') function fired");
  boardReady = true;
});
//--------------------------------------

Here's my code to send a test message across the i2c to my Arduino running at address 0x0c (12):

//--------------------------------------
    if (boardReady) {
      var write = (message) => {
        board.i2cWrite(0x0c, Array.from(message, c => c.charCodeAt(0)));
        console.log("5) i2c write actually executed.  Message: " + message);
      };
      console.log ("2) pre i2cConfig call");
      board.i2cConfig({address:'/dev/i2c-1'}); 
      console.log("3) pre repl inject call");
      board.repl.inject({ write });
      console.log("4) pre write() call");
      write("Hello World");
      console.log("6) calling callback");
      callback({success:true});
    }
//--------------------------------------

When executed, this generates the following output and error:

1) i2c open called.  boardReady: true
2) pre i2cConfig call
3) pre repl inject call
4) pre write() call
2018-03-16T14:50:33.164Z - error: Server Error  status: 500 message: Cannot read property 'i2cWriteSync' of undefined

So . . . from all of my log statements it is clear that the error occurs inside the Johnny Five code somewhere when I call i2cWrite().

Any help would be greatly appreciated.  I am dying to get i2c communication working.

Thanks!

fivdi, 2018-03-16:
I can't see any code that would output `"i2c open called. boardReady: true"`.

`'/dev/i2c-1'` is not a valid I2C address. An I2C address is a number.

If you would like to use the board I2C methods to talk to an I2C device you'll need to read up on how they work at https://github.com/rwaldron/io-plugins.

If an error was thrown please post the stack trace. Without a stack trace it's going to be difficult to figure out what's going on.

Please post a complete program rather only parts of the program.

fivdi, 2018-03-16:
It may be easier to use the [i2c-bus](https://github.com/fivdi/i2c-bus) module directly rather than the board I2C methods. BeagleBone-IO uses i2c-bus under the covers.

totem415, 2018-03-20:
Thanks for all of your help.  I've been out of town for a couple of days, but I'm getting back into it today.

That's interesting about Johnny 5 using i2c-bus.  Looked at that last week, but it doesn't seem to have any support (or it's not documented) for running as an i2c master.  Only as a slave.  I need to run a master that talks to an Arduino & possibly other slaves.

OK, so here's a standalone program that generates the error:

//------------------------------------------------------
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
  io: new BeagleBone()
});
 
board.on('ready', function() {
  console.log("1) johnny five board.on('ready') function fired");
  var write = (message) => {
      
    board.i2cWrite(0x0c, Array.from(message, c => c.charCodeAt(0)));  // ERROR ON THIS LINE
    
    console.log("3) i2c write actually executed.  Message: " + message);
  };
  board.i2cConfig({address:1});  
  board.repl.inject({ write });
  console.log("2) pre write() call");
  write("Hello World");
  console.log("4) post write() call");
});
//------------------------------------------------------

...here's a snapshot of my console:

root@1UpTester:/var/lib/cloud9/md/trunk# node i2cTest
1521583428327 Available BeagleBone-IO
1521583428472 Connected BeagleBone-IO
1521583428667 Repl Initialized
>> 1) johnny five board.on('ready') function fired
2) pre write() call
/var/lib/cloud9/md/trunk/node_modules/linux-io/lib/linux-io.js:357
    i2c.i2cWriteSync(address, buffer.length, buffer);
       ^

TypeError: Cannot read property 'i2cWriteSync' of undefined
    at BeagleBone.LinuxIO.i2cWrite (/var/lib/cloud9/md/trunk/node_modules/linux-io/lib/linux-io.js:357:8)
    at Board.(anonymous function) [as i2cWrite] (/var/lib/cloud9/md/trunk/node_modules/johnny-five/lib/board.js:484:21)
    at write (/var/lib/cloud9/md/trunk/i2cTest.js:12:11)
    at Board.<anonymous> (/var/lib/cloud9/MD/trunk/i2cTest.js:19:3)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickDomainCallback (internal/process/next_tick.js:128:9)

Also, here's some more info on the i2c bus on this Beaglebone Black Rev C running Debian 9.2:

ls -l /sys/bus/i2c/devices/i2c-*
OUTPUT:
lrwxrwxrwx 1 root root 0 Jan  1  2000 /sys/bus/i2c/devices/i2c-0 -> ../../../devices/platform/ocp/44e0b000.i2c/i2c-0
lrwxrwxrwx 1 root root 0 Jan  1  2000 /sys/bus/i2c/devices/i2c-1 -> ../../../devices/platform/ocp/4802a000.i2c/i2c-1
lrwxrwxrwx 1 root root 0 Jan  1  2000 /sys/bus/i2c/devices/i2c-2 -> ../../../devices/platform/ocp/4819c000.i2c/i2c-2

...and just in case:
USE THIS CHART TO CHECK MEM ADDRESSES TO KNOW WHICH PORT IS WHICH:
i2c0: 0x44E0_B000
i2c1: 0x4802_A000
i2c2: 0x4819_C000

If you have any idea what the problem is, please do share, and thanks.

-roger

fivdi, 2018-03-21:
>  I need to run a master that talks to an Arduino & possibly other slaves.

This is what i2c-bus supports.

Please change the following code:
```js
  board.i2cConfig({address:1});  
```
to:
```js
  board.i2cConfig({address:0x0c});  
```
and give it a try.

The address to specify when calling `i2cConfig` is the address of the device with which you would like to communicate.

totem415, 2018-03-21:
Booya!!
```
>> 1) johnny five board.on('ready') function fired
2) pre write() call
3) i2c write actually executed.  Message: Hello World
4) post write() call
```

And the serial monitor on the Arduino:
```
Arduino connected.  Listening for incoming I2C bytes.
Hello World
```

Thanks a million.  It's working.

closed 2018-03-21 by totem415