julianduque/beaglebone-io #65: PWM Frequency

badfrogg, 2018-03-19:
I've been working my way backwards through the source code from my BBB project that is using node-red-contrib-gpio. I've been trying to figure out how to manipulate the PWM frequency. I see in the begalebone.js code that looks like it is hard coded in there to be 2kHz. So regardless of what I do to try to change that at run time, if im using this lib then it will change it back to 2kHz every time it's called. Is that correct?

I'm very amateur at JS & github and not sure what I could do to fix this (for me). Do i fork this, change the value in the begalebone.js to something that will work for me and then recompile? do you you even compile JS, no right. I just have to figure out how to install my own fork... Am I at all on the right track?

Thanks for any direction.

fivdi, 2018-03-19:
@badfrogg Yes, your assumption is correct. The PWM frequency is hard coded to 2kHz [here](https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/beaglebone.js#L10). First and foremost BeagleBone-IO is a Johnny-Five IO Plugin, This means it implements methods specified in the [IO Plugins Specification](https://github.com/rwaldron/io-plugins). The specification doesn't require that it be possible to change the PWM frequency so BeagleBone-IO doesn't offer this functionality.

> I'm very amateur at JS & github and not sure what I could do to fix this (for me). Do i fork this, change the value in the begalebone.js to something that will work for me and then recompile? do you [...]

The easiest thing to do would be to install BeagleBone-IO with `npm install beaglebone-io` and then edit file node_modules/beaglebone-io/lib/beaglebone.js and manully change [the 2000 here](https://github.com/julianduque/beaglebone-io/blob/v3.0.0/lib/beaglebone.js#L10) to whatever it should be.  No recompile would be needed as only a JavaScript file would have been modified. I think it will be necessary to ensure that PWM_PERIOD is an integer.

If you need a different PWM frequency on different pins more work would be needed.

You could also fork this repository and modify it as needed. To install your fork run the command `npm install badfrogg/beaglebone-io`.

badfrogg, 2018-03-20:
I get all of that. This stuff is not that easy to teach yourself so its a big deal when someone takes the time to give some direction.

Thanks!

closed 2018-03-20 by fivdi