julianduque/beaglebone-io #66: GPIO error creating a new pin instance on a Beaglebone Black in Debian 9.2

totem415, 2018-04-04:
I'm on a Beaglebone Black, Debian 9.2, Node JS v6.11.4, running the latest beaglebone-io / johnny five combo.

I'm getting this error when I create a new pin instance to control a GPIO to use as a digital in.

Here's my standalone app code:
```
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
    io: new BeagleBone()
});

board.on("ready", function() {
    var mechRprtReadyPin = new five.Pin({
        pin: "P9_23",
        type: 'digital',
        mode: 0
    });
    mechRprtReadyPin.on('high', onMechRprtPinHigh);
    mechRprtReadyPin.on('low', onMechRprtPinLow);
    console.log("Pin Control initialization complete.");
});

function onMechRprtPinHigh(data) {
    console.log('PIN: HIGH (mechaduino report event)  data: ' + JSON.stringify(data));
}

function onMechRprtPinLow(data) {
    console.log('Pin: LOW  data: ' + JSON.stringify(data));
}
```
And here's the output / call stack of the error:
```
root@1UpTester:/var/lib/cloud9/md/trunk# node gpioTest
1522802125440 Available BeagleBone-IO
1522802125585 Connected BeagleBone-IO
1522802125782 Repl Initialized
>> fs.js:642
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
    at Error (native)
    at Object.fs.openSync (fs.js:642:18)
    at Object.fs.writeFileSync (fs.js:1348:33)
    at Object.module.exports.state (/var/lib/cloud9/md/trunk/node_modules/beaglebone-io/lib/pinmux.js:9:6)
    at BeagleBone._pinModeInput (/var/lib/cloud9/md/trunk/node_modules/beaglebone-io/lib/beaglebone.js:34:10)
    at BeagleBone.LinuxIO.pinMode (/var/lib/cloud9/md/trunk/node_modules/linux-io/lib/linux-io.js:99:12)
    at Pin.set (/var/lib/cloud9/md/trunk/node_modules/johnny-five/lib/pin.js:82:17)
    at new Pin (/var/lib/cloud9/md/trunk/node_modules/johnny-five/lib/pin.js:90:13)
    at Board.<anonymous> (/var/lib/cloud9/md/trunk/gpioTest.js:9:28)
    at emitNone (events.js:91:20)
```
The error is generated on this line:
```
var mechRprtReadyPin = new five.Pin({
        pin: "P9_23",
        type: 'digital',
        mode: 0
    });
```
Any help would be very much appreciated.

fivdi, 2018-04-04:
I looks like the BeagleBone Black is not configured correctly.

Please post the complete output of the following commands
```
cat /etc/dogtag 
cat /etc/debian_version 
cat /boot/uEnv.txt 
ls -l /sys/devices/platform/ocp/
ls -l /sys/devices/platform/ocp/ocp\:P9_23_pinmux/
sudo /opt/scripts/tools/version.sh
```

totem415, 2018-04-04:
Thanks very much for taking a look.  Here's my output:

```
root@1UpTester:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2017-10-10
root@1UpTester:~# cat /etc/debian_version
9.2
root@1UpTester:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.25-ti-r38
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
[... 76 lines cut ...]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]:[location: dd MBR]
kernel:[4.14.25-ti-r38]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1~stretch+20171009]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.901457] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.902558] gpio-of-helper ocp:cape-universal: ready
END
root@1UpTester:~#
```

fivdi, 2018-04-04:
The problem is most likely being caused by the following lines in `/boot/uEnv.txt`
```
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/BB-I2C1-00A0.dtbo
```
Please remove these lines from `/boot/uEnv.txt`, reboot the BeagleBone Black and try running BeagleBone-IO / Johnny-Five program again. It should work.

To enable a UART or I2C bus please used the config-pin utility.

For example, lets say I2C1_SCL should be on P9.17 and I2C1_SDA should be on P9.18. To perform this configuration the following commands need to be run:
```
config-pin P9.17 i2c
config-pin P9.18 i2c
```

RobertCNelson, 2018-04-04:
@totem415 couple things are going on here.

By selecting:
```
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/BB-I2C1-00A0.dtbo
```
in /boot/uEnv.txt you've actually disabled "cape-universal" for v4.14.x (technically: v4.4.x->v4.14.x)

This is why this failed:
```
Error: ENOENT: no such file or directory, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
```

So to make this work. (i assumed users wouldn't do what you did), you should comment out all your:

uboot_overlay_addrX options and enable the pinmux's via config-pin..

BUT... (i assumed) so i have work in progress to allow what you tried to do actually work.. (this requires v4.16.x)

1: fix pru overlay (v4.4.x-ti wont work on v4.14.x/v4.16.x)
```
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
```
In /boot/uEnv.txt change uboot_overlay_pru:
```
from:
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
to:
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
```
2: Install/build git version of bb.org-overlays
(i pushed this patch just for you: https://github.com/beagleboard/bb.org-overlays/commit/2aa90ed308c0f8ce4cbc7b1e2653875ed888ef9c )

```
debian@test-bbb-2:/opt/source/bb.org-overlays$ git pull
debian@test-bbb-2:/opt/source/bb.org-overlays$ make
debian@test-bbb-2:/opt/source/bb.org-overlays$ sudo make install
```
3: upgrade the bootloader: (this is a new feature of my 2018.03 version)

```
debian@test-bbb-2:~$ sudo /opt/scripts/tools/developers/update_bootloader.sh

```
4: install v4.16.0 based kernel... which is still building...

http://gfnd.rcn-ee.org:81/farm/deb/

```
debian@test-bbb-2:~$ sudo /opt/scripts/tools/update_kernel.sh --bone-kernel --exp
```
only when "4.16.0-bone7" is done, as "4.16.0-rc7-bone7" is broken..

```
debian@test-bbb-2:~$ sudo /opt/scripts/tools/update_kernel.sh --bone-kernel --exp
info: checking archive
2018-04-04 21:59:49 URL:https://rcn-ee.com/repos/latest/stretch-armhf/LATEST-omap-psp [198/198] -> "LATEST-omap-psp" [1]
-----------------------------
Kernel Options:
ABI:1 LTS41 4.1.38-bone24
ABI:1 LTS44 4.4.126-bone22
ABI:1 LTS49 4.9.91-bone10
ABI:1 LTS414 4.14.32-bone14
ABI:1 STABLE 3.8.13-bone86
ABI:1 TESTING 4.15.15-bone4
ABI:1 EXPERIMENTAL 4.16.0-rc7-bone7
-----------------------------
Kernel version options:
-----------------------------
LTS44: --lts-4_4
LTS49: --lts-4_9
LTS414: --lts-4_14
STABLE: --stable
TESTING: --testing
-----------------------------
info: you are running: [4.9.88-ti-r107], latest is: [4.16.0-rc7-bone7] updating...
Get:1 http://repos.rcn-ee.com/debian stretch InRelease [3,058 B]
Ign:2 http://deb.debian.org/debian stretch InRelease 
^C
```

Regards,

totem415, 2018-04-04:
Wow.  @fivdi , you're right.  That is amazing!  Thank you for all of your help.

@RobertCNelson  Thank you so much for looking into it and pushing a fix so that our configuration will work.  I'll keep my eyes peeled for your new kernel release.

RobertCNelson, 2018-04-05:
@totem415 just checked this morning, it looks like everything works:

```
sudo /opt/scripts/tools/update_kernel.sh --bone-kernel --exp
```

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[70edebd65fe6ea7de16ea8efe3c48b2a4062034a]
eeprom:[A335BNLT000C1727BBBG1071]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-g254339602c]:[location: dd MBR]
kernel:[4.16.0-bone7]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-UART5-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180404.0-0rcnee0~stretch+20180404]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0]
dmesg | grep pinctrl-single
[    0.906981] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.919706] gpio-of-helper ocp:cape-universal: ready
END
debian@beaglebone:~$ ls /sys/devices/platform/ocp/ | grep pinmux
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_13_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_19_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_12_pinmux
ocp:P9_14_pinmux
ocp:P9_15_pinmux
ocp:P9_16_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_25_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
debian@beaglebone:~$ 
```

PS, don't let the "--exp" scare you, in a week or two, the "v4.16.x" will be moved to "--testing", (we have "3.8.13" stuck endlessly in --stable) so --testing is really our stable channel for the bones..

Regards,

totem415, 2018-04-06:
OK, so that seems to get me past my initialization error.  Fantastic!

I'll report back in the morning when I've done some more testing to make sure everything is working correctly, but this is looking great.  

Thank you so much!!

totem415, 2018-04-06:
Confirmed.  The GPIO is working.  

This experience really gives me a lot more confidence using the Beaglebone-IO / Johnny Five combo as it is clearly well supported.

Thanks again, Robert.

-roger

totem415, 2018-04-13:
Hi.

Sorry to bother you again, but I've moved to a different Beaglebone Black Rev C, but I can't get the i2c to work even though I've followed all of the steps that you outlined to update uEnv.txt and all of the other steps.

I'm wondering if the safest thing to do is to go back to your original idea to fix the problem:

"So to make this work. (i assumed users wouldn't do what you did), you should comment out all your:  uboot_overlay_addrX options and enable the pinmux's via config-pin.."

Any thoughts are appreciated.  Thanks.

Here's my output on the new board:
```
root@1UpTester:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2017-10-10
root@1UpTester:~# cat /etc/debian_version
9.2
root@1UpTester:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.16.0-bone7
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet
[... 99 lines cut ...]
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P8_45_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P8_46_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_12_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_14_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_15_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_16_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_19_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_20_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_23_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_25_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_27_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_28_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_29_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_30_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_31_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_41_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_42_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_91_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_92_pinmux
lrwxrwxrwx 1 root root    0 Apr 13 19:36 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Apr 13 19:36 power
lrwxrwxrwx 1 root root    0 Jan  1  2000 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Jan  1  2000 uevent
root@1UpTester:~#
```

RobertCNelson, 2018-04-13:
@totem415 let's see the important bits:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

fivdi, 2018-04-13:
@totem415 Also, which I2C bus would you like to use?

The following line from `uEnv.txt` would indicate that you would like to use I2C1.
```
uboot_overlay_addr4=/lib/firmware/BB-I2C1-00A0.dtbo
```

However, BeagleBone-IO defaults to using I2C2. It won't be possible to configure BeagleBone-IO to use anything other that I2C2 until #60 is fixed.

fivdi, 2018-04-13:
@totem415 please answer the question that was asked [here](https://github.com/julianduque/beaglebone-io/issues/66#issuecomment-381249088).

totem415, 2018-04-13:
Sorry.

Yes, i2c 2, via pins P9_19 and P9_20.

Should I change that line to I2C2 to fix?

Thanks!

-roger

fivdi, 2018-04-13:
If you're not using I2C1 that line shouldn't be needed.

What's the output of command:
```
i2cdetect -y -r 2
```
Is the address of the I2C device that you would like to communicate with visible in the output of the command?

If the address is not visible the I2C device probably isn't connected to the to the bus correctly.

fivdi, 2018-04-13:
Are you really using all the UARTs?

UART1 may be using P9_18 and P9_19 for RTS and CTS. I can't remember where the source code for the device tree overlays are to check.

Remove the following line from `uEnv.txt`, reboot and try again.

```
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
```

totem415, 2018-04-13:
Don't need all the UARTs.  Migrating away from the one we were using, so none atm, but we might want a couple down the road.  Not picky about which ones.

Thanks!

fivdi, 2018-04-13:
Is it working now?

RobertCNelson, 2018-04-13:
RTC/CTS has a special overlay names ;)

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART2-RTSCTS-00A0.dts

Regards,

fivdi, 2018-04-13:
Yes. UART1 doesn't use P9.19 and P9.20. See https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART1-00A0.dts#L28-L29. Sorry for the confusion.

In addition to the device at address 12 being missing, the output of `i2cdetect -y -r 2` doesn't look right  to me. I'd expect to see:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

```

totem415, 2018-04-13:
Yeah, I remember seeing all of those 'UU' lines in there before when it was working on the other board.

OK, so changing my uEnv.txt line to:
```
uboot_overlay_addr4=/lib/firmware/BB-I2C2-00A0.dtbo
```
...did not fix the problem.  Same result as before.

RobertCNelson, 2018-04-13:
The UU's are when the kernel has a driver connected to that address.

Are you sure the avr is in i2c slave mode?

Regards,

totem415, 2018-04-13:
The avr?  Do you mean the Arduino?  That's definitely in slave mode and worked beautifully before.  The BBB we are configuring here should be the master.

-rog

fivdi, 2018-04-13:
Yes, the Arduino.
Are the I2C bus SDA and SCL lines connected the right way around?
Are you sure they are not reversed?

totem415, 2018-04-13:
I double-checked and they are wired up correctly.  I suppose it might be a bad wire, though.  But even if a wire was bad, wouldn't I still get all of those "UU" lines that show up for:
```
i2cdetect -y -r 2
```
?

RobertCNelson, 2018-04-13:
@totem415 no, "UU" = kernel had loaded a driver at that address and has locked it... Thus normal user can not access..  In your case i'd expect "12" based on your messages..

Regards,

totem415, 2018-04-13:
OK, I'll try replacing the wires after I get a bite.  Will report back.

Thanks again for looking at this.

-roger

totem415, 2018-04-14:
I replaced the wires.  "12" is still not showing up.  Also the wires I replaced tested out ok, so I don't think it's bad wiring.

fivdi, 2018-04-14:
Are there pullup resistors connecting SDA and SCL to 3.3V?

totem415, 2018-04-16:
Pullup resistors?  No.  Just plain wires.  I wasn't using them before, though, and it worked beautifully.

RobertCNelson, 2018-04-16:
https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-I2C1-00A0.dts#L47
```
BONE_P9_18 0x72 -> 0b1110010
```
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/dt-bindings/pinctrl/omap.h?h=v4.17-rc1
```
#define PULL_ENA		(1 << 3)
#define PULL_UP			(1 << 4)
``
change those two lines to:

```
BONE_P9_18 (PIN_INPUT_PULLUP | MUX_MODE2)
BONE_P9_17 (PIN_INPUT_PULLUP | MUX_MODE2)
```
they are under /opt/source/bb.org-overlays/
```
cd /opt/source/bb.org-overlays/
git pull
#edit src/arm/BB-I2C1-00A0.dtss
make
sudo make install
sudo reboot
```
Regards,

totem415, 2018-04-16:
Hi Robert.  Thanks for sharing your advice.  I have a couple of questions just to make sure we're all on the same page.

I am wired up to use i2c bus 2 (on pins 19 & 20).  I know there was some confusion in the thread, but it looks like you are configuring the BBB to use i2c 1.  Please confirm that this will fix access to i2c bus 2 (on pins 19 & 20). 

It looks like your configuration would enable i2c bus 1.  Yes?  How do I do this for Bus 2 (pins 19 & 20)?

Also, there doesn't appear to be a file at:
```
/opt/source/bb.org-overlays/src/arm/BB-12C1-00A0.dtss
```
Here are the closest matches in the same directory:
```
BB-I2C1-00A0.dtbo
BB-I2C1-00A0.dts
```
Please advise . . . and thanks again so much for the help!

-roger

totem415, 2018-04-18:
@RobertCNelson I'd really love to get a response from you to my question.

fivdi, 2018-04-18:
@totem415 There was confusion with I2C1 and I2C2. Normally I2C2 is available and configured for usage after a BeagleBone Black is booted and normally no additional steps need to be performed to use I2C2. Have you tried with external pullup resistors that connect SDA and SCL to 3.3V?

fivdi, 2018-04-19:
Moved to https://github.com/julianduque/beaglebone-io/issues/71

@totem415 Please don't reopen issues that have already been closed to report new issues. I would imagine that was one of the causes of confusion here.

closed 2018-04-19 by fivdi

totem415, 2018-04-19:
Thank you fivdi.

And sorry for reopening the old ticket.  

Will give the pullup resistors a shot.  And if that doesn't get it, I'll probably just start with a fresh Debian 9.2 flash.  It was the UARTs that led us into configuration in the first place, and we don't even need that any more.  

Will report back.

totem415, 2018-04-20:
No go. 

I used 5K pullup resistors to connect both sides to 3.3V.  No change in the function.  My next step was to start with a fresh flash of Debian 9.2.  It shows the i2c bus as being there, but I don't see my device showing up. 

Will leave this thread closed, close the other I started yesterday, and start a fresh one (simplest yet) with my current situation.

Thanks again!