julianduque/beaglebone-io #71: Unable to configure Debian 9.2 on a Beaglebone Black to use i2c with beaglebone-io and Jonny Five

totem415, 2018-04-18:
I'm trying to configure my system to use i2c with beaglebone-io and Johnny Five node modules.

This issue is stemming from another thread that seems to have gotten its water very muddy with different stuff.  So I'm starting fresh with my current situation.  Also I'm getting a little desperate to fix this, so I thought I'd put all of the latest info in one place.

I'm trying to activate i2c bus 2 (pins p9_19 and p9_20).  This worked on a previous beagle bone black system but I have not been able to correctly configure the new system to work where i can see an arduino I have attached and set up as a slave at address 0x0c (12).

Formerly, I was attempting to make this line work in uEnv.txt:
```
uboot_overlay_addr4=/lib/firmware/BB-I2C2-00A0.dtbo
```
But I've switched over to using these:
```
config-pin P9.19 i2c
config-pin P9.20 i2c
```
...though I've tried both and neither one seems to work.

I am running Debian 9.2.  

I have modified:
```
/opt/source/bb.org-overlays/src/arm/BB-12C2-00A0.dts
```
as follows:
```
        fragment@0 {
                target = <&ocp>;
                __overlay__ {
                        P9_19_pinmux { status = "i2c2_scl"; };  /* disabled or  i2c2_scl */
                        P9_20_pinmux { status = "i2c2_sda"; };  /* disabled or  i2c2_sda */
                };
        };

        fragment@1 {
                target = <&am33xx_pinmux>;
                __overlay__ {
                        bb_i2c2_pins: pinmux_bb_i2c2_pins {
                                pinctrl-single,pins = <
                                        BONE_P9_20 0x73 (PIN_INPUT_PULLUP | MUX_MODE2)  /* i2c2_sda, SLEWCTRL_SLOW | INPUT_PULLUP | MODE3 */
                                        BONE_P9_19 0x73 (PIN_INPUT_PULLUP | MUX_MODE2)  /* i2c2_scl, SLEWCTRL_SLOW | INPUT_PULLUP | MODE3 */
                                >;
                        };
                };
        };
```

Here's a bunch more info about my system:
```
root@1UpTester:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2017-10-10
root@1UpTester:~# cat /etc/debian_version
9.2
root@1UpTester:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.16.0-bone7
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
[... 189 lines cut ...]
root@1UpTester:~# dmesg | grep i2c
[    0.501649] omap_i2c 44e0b000.i2c: could not find pctldev for node /ocp/l4_wkup@44c00000/scm@210000/pinmux@800/pinmux_i2c0_pins, deferring probe
[    0.503106] omap_i2c 4819c000.i2c: bus 2 rev0.11 at 100 kHz
[    1.938758] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[    2.946752] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[    3.954750] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[    4.962750] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[    5.122484] i2c /dev entries driver
[    5.293492] input: tps65217_pwr_but as /devices/platform/ocp/44e0b000.i2c/i2c-0/0-0024/tps65217-pwrbutton/input/input0
[    5.295097] omap_i2c 44e0b000.i2c: bus 0 rev0.11 at 400 kHz
[  126.211695] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  127.219510] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  128.227326] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  129.235147] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  130.242963] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  131.250790] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  132.258632] omap_i2c 4819c000.i2c: timeout waiting for bus ready
[  133.266480] omap_i2c 4819c000.i2c: timeout waiting for bus ready
```

Any advice is greatly appreciated.  I'm not trying to do anything fancy on this system in terms of hardware, I just need a bunch of GPIO pins and i2c bus 2 (P9_19 & P9_20).

Thanks!

-roger

fivdi, 2018-04-19:
I'd suggest going back to square one and taking things from there.

* Flash image `"Debian 9.3 2018-03-05 4GB SD IoT"` or `"Debian 9.3 2018-01-28 4GB SD LXQT"` for the BeagleBone Black available [here](https://beagleboard.org/latest-images) onto an SD card
* Boot the BeagleBone from the SD card (hold down the boot button during power on and keep the boot button pressed till all four LEDS turn on at the same time, keep the boot button pressed as long as the four LEDS are on)
* Don't modify anything in `/config/boot.txt`
* Don't upgrade anything
* Check to see if your Johnny-Five / BeagleBone-IO program works. It should.

totem415, 2018-04-26:
Hi Fivdi.

OK, sorry I moved to Debian 9.3 as you mentioned, no modification  whatsoever to uEnv.txt and I still get the same problem.

Here's my Node JS program:
```
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
    io: new BeagleBone()
});
 
var write;
board.on('ready', function() {
    write = (message) => {
        board.i2cWrite(0x0c, Array.from(message, c => c.charCodeAt(0))); 
    };
    board.i2cConfig({address:0x0c});  
    board.repl.inject({ write });
    write("first message");
});
```
Here's my error:
```
root@beaglebone:/var/lib/cloud9/MercuryDesign/trunk# node i2cTest2
1524706314862 Available BeagleBone-IO
1524706314933 Connected BeagleBone-IO
1524706314994 Repl Initialized
>> fs.js:796
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EBUSY: resource busy or locked, write
    at Error (native)
    at Object.fs.writeSync (fs.js:796:20)
    at Bus.i2cWriteSync (/var/lib/cloud9/MercuryDesign/trunk/node_modules/i2c-bus/i2c-bus.js:348:13)
    at BeagleBone.LinuxIO.i2cWrite (/var/lib/cloud9/MercuryDesign/trunk/node_modules/linux-io/lib/linux-io.js:357:9)
    at Board.(anonymous function) [as i2cWrite] (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/board.js:502:21)
    at write (/var/lib/cloud9/MercuryDesign/trunk/i2cTest2.js:12:15)
    at Board.<anonymous> (/var/lib/cloud9/MercuryDesign/trunk/i2cTest2.js:17:5)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
```

I'm now hooked up to the Arduino with 50 ohm pullup resistors.  No dice.  I've already replaced all of the wiring and tested the old . . . it was working.

totem415, 2018-04-26:
One more thing.

Just as a reality check, I change the address of my i2c device  to 0x0b (wrong address) and it's very interesting that I get a different error message when I run that:
```
root@beaglebone:/var/lib/cloud9/MercuryDesign/trunk# node i2cTest2
1524707437946 Available BeagleBone-IO
1524707438017 Connected BeagleBone-IO
1524707438079 Repl Initialized
>> /var/lib/cloud9/MercuryDesign/trunk/node_modules/linux-io/lib/linux-io.js:357
    i2c.i2cWriteSync(address, buffer.length, buffer);
       ^

TypeError: Cannot read property 'i2cWriteSync' of undefined
    at BeagleBone.LinuxIO.i2cWrite (/var/lib/cloud9/MercuryDesign/trunk/node_modules/linux-io/lib/linux-io.js:357:8)
    at Board.(anonymous function) [as i2cWrite] (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/board.js:502:21)
    at write (/var/lib/cloud9/MercuryDesign/trunk/i2cTest2.js:13:15)
    at Board.<anonymous> (/var/lib/cloud9/MercuryDesign/trunk/i2cTest2.js:18:5)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickDomainCallback (internal/process/next_tick.js:128:9)
```

So it seems that it's getting farther (and a different error) when you give it the correct address which implies that it's actually connected and trying to communicate.

One more piece of info:
```
oot@beaglebone:/var/lib/cloud9/MercuryDesign/trunk# i2cdetect -r 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@beaglebone:/var/lib/cloud9/MercuryDesign/trunk#
```
The device at 0x0c isn't appearing here either.

fivdi, 2018-04-26:
Please post the Arduino sketch.

fivdi, 2018-04-26:
Also, please post the output of
```
sudo /opt/scripts/tools/version.sh
cat /boot/uEnv.txt
ls -l /sys/devices/platform/ocp/
```

totem415, 2018-04-26:
Thanks a million for taking a look, @fivdi.

Arduino Sketch:
```
#include "Arduino.h"
#include <string.h>
#include <stdio.h>
#include <Wire.h>
//#include "I2C_Anything.h"

// this is the i2c 7-bit bus address assignment for a slave:
#define CTRL_I2C_ADDR 11  

// i2c variables:
const int maxlength = 72;
char buffer[maxlength];
char printable[maxlength];
int received = 0;
char i2cCmdResponse[maxlength];
const int RESP_BYTES_COUNT = 64;
const int REPORT_IMPULSE_PIN = 2;

void setup() {
  digitalWrite(13,HIGH);  //13 is the led pin on the Arduino Uno
  
  Serial.begin(115200);   // RAC: Mechauino setup for Serial instead of Serial (for Serial Monitor)
  delay(3000); 
  //i2c:
  Wire.begin(CTRL_I2C_ADDR);       // join i2c bus with address #11 or 0x0b
  Wire.onReceive(receiveI2CEvent); // register event
  Wire.onRequest(requestI2CEvent); // register event
  pinMode(REPORT_IMPULSE_PIN, OUTPUT); 
  
  Serial.println("i2c tester initialization complete.  Listening for i2c events over the wire.");
  digitalWrite(13, LOW);
}

void loop() {
}

// Message from i2c Master (no response)
void receiveI2CEvent(int howMany) {
  received = howMany;
  memset(buffer, 0, maxlength);
  for (int i = 0; i < howMany; i++) {
    buffer[i] = Wire.read();
  }
  if (received > 0) {
    memcpy(printable, buffer, maxlength);
    for (int i = 0; i < received; i++) {
      Serial.print(printable[i]);
    }
    Serial.println(""); 
    received = 0;  
  }
}

// Requested bytes from i2c Master (send response)
void requestI2CEvent() {
  Serial.println("requestEvent() called");
  Wire.write("thanks");  // 6 characters
}
```
root@1upTester:/home/debian# /opt/scripts/tools/version.sh
git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT00C05014BBBK09A6]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
[... 143 lines cut ...]
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_12_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_13_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_14_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_15_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_16_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_17_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_18_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_19_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_20_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_21_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_22_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_23_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_24_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_26_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_27_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_30_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_41_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_42_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_91_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_92_pinmux
lrwxrwxrwx 1 root root    0 Apr 26 20:13 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Apr 26 20:13 power
lrwxrwxrwx 1 root root    0 Jan  1  2000 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Jan  1  2000 uevent
```

fivdi, 2018-04-26:
I tried your code out and everything works correctly for me.
```
debian@beaglebone:~/beaglebone-io$ i2cdetect -y -r 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
```js
debian@beaglebone:~/beaglebone-io$ cat t.js
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
    io: new BeagleBone()
});
 
var write;
board.on('ready', function() {
    write = (message) => {
        board.i2cWrite(0x0b, Array.from(message, c => c.charCodeAt(0))); 
    };
    board.i2cConfig({address:0x0b});  
    board.repl.inject({ write });
    write("first message");
});
```
```
debian@beaglebone:~/beaglebone-io$ node t.js
1524778408291 Available BeagleBone-IO  
1524778408345 Connected BeagleBone-IO  
1524778408416 Repl Initialized  
>> 
```
The output on the Arduino serial monitor:
```
requestEvent() called
first message
```
Note that it was necessary to use external 4.7K pullup resistors on SDA and SCL. (Pullup to 3.3V.) It's didn't work without the external pullups. Also, don't forget to connect 0V on the Arduino and BeagleBone Black.

totem415, 2018-04-26:
No kidding.

I'm currently using  5K resistor (all I had around)  but I figure it's got to be close enough.

Regarding the 0V, do you mean I should connect the ground pins on the 2 boards to each other?

Thanks so much for taking the time to do this.  Maybe I have a bad wire or something?  Not sure, but it gives me some stuff to try.

-roger

fivdi, 2018-04-26:
> I'm currently using 5K resistor (all I had around) but I figure it's got to be close enough.

Correct.

> Regarding the 0V, do you mean I should connect the ground pins on the 2 boards to each other?

Yes.

totem415, 2018-04-27:
Hi @fivdi .

OK, so after fooling around with the wires a bit I did get the thing working.  Yay!  Closing ticket.

Sadly I haven't been able to get the original Arduino Zero to show up yet.  I'm still replacing wires and such, but clearly at this point it's not beaglebone-io's fault.

Thank you so much for helping me out with this, @fivdi .  You are really awesome.

-roger

closed 2018-04-27 by totem415