julianduque/beaglebone-io #72: Error: ENOENT: no such file or directory, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'

totem415, 2018-06-13:
Weird.

I'm running off of a uSD card that I was using on my last Beaglebone Black, so I would assume all files in the file system haven't changed.  Yet, I'm getting this new error that I wasn't getting before.  I moved my uSD card to a new Beaglebone Black Rev C (just like the last that fried on me).  Everything was working before with the default install of Debian 9.3 . . . no extra configuration required.  But on the new beaglebone I get this:
```
fs.js:642
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
    at Error (native)
    at Object.fs.openSync (fs.js:642:18)
    at Object.fs.writeFileSync (fs.js:1356:33)
    at Object.module.exports.state (/var/lib/cloud9/MercuryDesign/trunk/node_modules/beaglebone-io/lib/pinmux.js:9:6)
    at BeagleBone._pinModeInput (/var/lib/cloud9/MercuryDesign/trunk/node_modules/beaglebone-io/lib/beaglebone.js:34:10)
    at BeagleBone.LinuxIO.pinMode (/var/lib/cloud9/MercuryDesign/trunk/node_modules/linux-io/lib/linux-io.js:99:12)
    at Pin.set (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/pin.js:82:17)
    at new Pin (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/pin.js:90:13)
    at Board.<anonymous> (/var/lib/cloud9/MercuryDesign/trunk/model/PinControl.js:27:32)
    at emitNone (events.js:91:20)
```

The error is generated on this line:
```
        var mechRprtReadyPin = new five.Pin({
            pin: "P9_23",
            type: 'digital',
            mode: 0
        });
```
And here's the usual diagnostic stuff:
```
root@1upTester:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2018-03-05
root@1upTester:~# cat /etc/debian_version
9.3
root@1upTester:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.82-ti-r102
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

root@1upTester:~# ls -l /sys/devices/platform/ocp/
total 0
drwxr-xr-x 3 root root    0 Jan  1  2000 40300000.ocmcram
drwxrwxr-x 5 root gpio    0 Jan  1  2000 44e07000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 44e09000.serial
drwxr-xr-x 4 root root    0 Jan  1  2000 44e0b000.i2c
drwxr-xr-x 5 root root    0 Jan  1  2000 44e35000.wdt
drwxr-xr-x 4 root root    0 Jan  1  2000 44e3e000.rtc
drwxr-xr-x 9 root root    0 Jan  1  2000 47400000.usb
drwxr-xr-x 3 root root    0 Jan  1  2000 48038000.mcasp
drwxr-xr-x 3 root root    0 Jan  1  2000 48042000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48044000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48046000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 48048000.timer
drwxr-xr-x 3 root root    0 Jan  1  2000 4804a000.timer
drwxrwxr-x 5 root gpio    0 Jan  1  2000 4804c000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 48060000.mmc
drwxr-xr-x 4 root root    0 Jan  1  2000 480c8000.mailbox
drwxr-xr-x 3 root root    0 Jan  1  2000 480ca000.spinlock
drwxr-xr-x 4 root root    0 Jan  1  2000 4819c000.i2c
drwxrwxr-x 5 root gpio    0 Jan  1  2000 481ac000.gpio
drwxrwxr-x 5 root gpio    0 Jan  1  2000 481ae000.gpio
drwxr-xr-x 4 root root    0 Jan  1  2000 481d8000.mmc
drwxr-xr-x 5 root root    0 Jan  1  2000 4830e000.lcdc
drwxr-xr-x 3 root root    0 Jan  1  2000 48310000.rng
drwxr-xr-x 4 root root    0 Jan  1  2000 49000000.edma
drwxr-xr-x 3 root root    0 Jan  1  2000 49800000.tptc
drwxr-xr-x 3 root root    0 Jan  1  2000 49900000.tptc
drwxr-xr-x 3 root root    0 Jan  1  2000 49a00000.tptc
drwxr-xr-x 7 root root    0 Jan  1  2000 4a100000.ethernet
drwxr-xr-x 4 root root    0 Jan  1  2000 4a326000.pruss-soc-bus
drwxr-xr-x 3 root root    0 Jan  1  2000 4c000000.emif
drwxr-xr-x 3 root root    0 Jan  1  2000 53100000.sham
drwxr-xr-x 3 root root    0 Jan  1  2000 53500000.aes
drwxr-xr-x 3 root root    0 Jan  1  2000 56000000.sgx
-rw-r--r-- 1 root root 4096 Jun 13 00:41 driver_override
-r--r--r-- 1 root root 4096 Jun 13 00:41 modalias
drwxr-xr-x 3 root root    0 Jan  1  2000 ocp:cape-universal
drwxr-xr-x 6 root root    0 Jan  1  2000 ocp:l4_wkup@44c00000
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_19_pinmux
drwxrwxr-x 3 root gpio    0 Jan  1  2000 ocp:P9_20_pinmux
lrwxrwxrwx 1 root root    0 Jun 13 00:41 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x 2 root root    0 Jun 13 00:41 power
lrwxrwxrwx 1 root root    0 Jan  1  2000 subsystem -> ../../../bus/platform
-rw-r--r-- 1 root root 4096 Jan  1  2000 uevent
root@1upTester:~# ls -l /sys/devices/platform/ocp/ocp\:P9_23_pinmux/
ls: cannot access '/sys/devices/platform/ocp/ocp:P9_23_pinmux/': No such file or directory
root@1upTester:~# sudo /opt/scripts/tools/version.sh
sudo: unable to resolve host 1upTester
git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT00C01618BBBK0525]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180305.0-0rcnee0~stretch+20180305]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.372401] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.373417] gpio-of-helper ocp:cape-universal: ready
END
```
Any help is greatly appreciated.

RobertCNelson, 2018-06-13:
@totem415 this will cause an issue:

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]:[location: dd MBR]
k
```

Run:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

To fix that..

Regards,

totem415, 2018-06-13:
Success!

Thanks very much @RobertCNelson .

You fixed my problem, so don't feel obligated to answer my question, but just in case...

...do you know why this happened?  I was under the impression that if you switch the uSD card from one Beaglebone to another and boot it with the boot button pressed (so it boots off the uSD card) that you'd have an EXACTLY similar file system and that it should work just like it did in the last one.  Any idea why this happened?  I assume I'll have to perform this same fix again if I migrate the uSD card to a new Beaglebone Black in the future?  I'm just trying to understand how this works.

Thanks again for your help, Robert.  Closing the issue.

-roger

closed 2018-06-13 by totem415