julianduque/beaglebone-io #73: Beaglebone-io appears to crash when run as a service

totem415, 2018-06-22:
Hi there.

I'm having an issue with my application repeatedly crashing after it launches as a service, initializes the hardware through beaglebone-io / Johnny Five and everything appears to be working with the GPIO.  Immediately after hardware initialization it gets the error "Board closing" and then the service is stopped and restarted.  I'm using the Express framework in Node JS, btw.  Unfortunately when this happens, it's not giving me a call stack, so I don't have that to pass along.  This application has worked perfectly in the past when run as a service, but not since we migrated from Linux 7.11 to 9.3 and changed to the Beaglebone-io node module.

This application works perfectly when launched via:
```
npm start
```  

The error happens when launched as a service.  It will repeatedly try to launch, then fail, relaunch, fail, etc.  Here is an example excerpt of what I'm seeing in the syslog, starting with my own application's log statements for the hardware finishing setup and appearing to work correctly:
```
Jun 22 20:00:07 1upTester Express[2582]: Pin: LOW
Jun 22 20:00:07 1upTester Express[2582]: uutPresenceDetect: Presence Pin went LOW
Jun 22 20:00:07 1upTester Express[2582]: uutPresenceDetect: Lid Pin went LOW
Jun 22 20:00:07 1upTester Express[2582]: 1529697607205 Board Closing.
Jun 22 20:00:17 1upTester systemd[1]: express.service: Service hold-off time over, scheduling restart.
Jun 22 20:00:17 1upTester systemd[1]: Stopped Express.
Jun 22 20:00:17 1upTester systemd[1]: Started Express.
Jun 22 20:00:21 1upTester Express[2596]: 1529697621840 Available BeagleBone-IO
```

This line before the restart is initiated seems like what failed:

```
Jun 22 20:00:07 1upTester Express[2582]: 1529697607205 Board Closing.
```
I'm running on the basic install of Debian 9.3 with Node JS version v6.13.0.

Any ideas what could be going on? 

Any help is appreciated.

-roger

RobertCNelson, 2018-06-22:
in your service file, bump up the value of RestartSec=, looks like the service timed out after 10 seconds..

Regards,

totem415, 2018-06-22:
Will try.  Thanks!

totem415, 2018-06-22:
OK, I gave it a shot at 20.  The same thing happens, however now it sits at the "Board Closing" message for 10 seconds before initiating restart:
```
Jun 22 20:45:37 1upTester Express[2777]: setting pin.  opts.addr: undefined opts.pin: P9_27
Jun 22 20:45:37 1upTester Express[2777]: deviceMgr:  uut presence detection initialized. success: true
Jun 22 20:45:37 1upTester Express[2777]: Pin: LOW
Jun 22 20:45:37 1upTester Express[2777]: uutPresenceDetect: Presence Pin went LOW
Jun 22 20:45:37 1upTester Express[2777]: uutPresenceDetect: Lid Pin went LOW
Jun 22 20:45:37 1upTester Express[2777]: 1529700337663 Board Closing.
```

When I run with:
```
npm start
```
...there is no "Board Closing" message and there are other hardware related log statements that happen when LEDs are turned off after a timer function... so it looks like my application is crashing at the "Board Closing" message when run as a service... and it's the Express Service's crash detection that is causing the restart.

Here is an excerpt of what I see when launched with 'npm start':
```
deviceMgr:  uut presence detection initialized. success: true
Pin: LOW
uutPresenceDetect: Presence Pin went LOW
uutPresenceDetect: Lid Pin went LOW
ledControl LED OFF at: P8_7
ledControl LED OFF at: P8_9
ledControl LED OFF at: P8_11
ledControl LED OFF at: P8_8
```

What do you think?

-roger

PS.  Thanks again for taking a look.

RobertCNelson, 2018-06-22:
Are you using full path's and exporting NODE_PATH, and systemd might be running it as un-privledged user, you can use the User= (uid) value in the service file to workaround that.

totem415, 2018-06-22:
Full path, yes minus the file extension.  That part definitely seems to work, though, because the application launches.

I'm not sure what you mean by "exporting NODE_PATH".  Do you mean putting something in my package.json for my node app?

If I log in as user "debian" without root privileges, I can launch the app and it works fine.

Here is my express.service file:
```
[Unit]
Description=Express
# Set dependencies to other services (optional)
#After=mongodb.service

[Service]
# Run Grunt before starting the server (optional)
#ExecStartPre=/usr/bin/grunt

# Start the js-file starting the express server
ExecStart=/var/lib/cloud9/MercuryDesign/trunk/bin/www
WorkingDirectory=/var/lib/cloud9/MercuryDesign/trunk
Restart=always
RestartSec=20
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=Express
# Change to a non-root user (optional, but recommended)
#User=<alternate user>
#Group=<alternate group>
# Set environment options
Environment=NODE_ENV=production #PORT=8080
PORT=8888

[Install]
WantedBy=multi-user.target
```

Thank you for looking at this.

-roger

totem415, 2018-06-26:
@RobertCNelson I did what you said and added a new user with sudo privelages and I specified that user in the express.service file as you suggested.

Interestingly, now when I run my program as a service, I get a different error with a call stack:
```
Jun 26 00:09:29 1upTester Express[1929]: Pin Control initialization complete.
Jun 26 00:09:29 1upTester Express[1929]: deiceMgr: pin control initialized.  Success: true
Jun 26 00:09:29 1upTester Express[1929]: setting pin.  opts.addr: undefined opts.pin: P9_23
Jun 26 00:09:29 1upTester Express[1929]: fs.js:642
Jun 26 00:09:29 1upTester Express[1929]:   return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
Jun 26 00:09:29 1upTester Express[1929]:                  ^
Jun 26 00:09:29 1upTester Express[1929]: Error: EACCES: permission denied, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
Jun 26 00:09:29 1upTester Express[1929]:     at Error (native)
Jun 26 00:09:29 1upTester Express[1929]:     at Object.fs.openSync (fs.js:642:18)
Jun 26 00:09:29 1upTester Express[1929]:     at Object.fs.writeFileSync (fs.js:1356:33)
Jun 26 00:09:29 1upTester Express[1929]:     at Object.module.exports.state (/var/lib/cloud9/MercuryDesign/trunk/node_modules/beaglebone-io/lib/pinmux.js:9:6)
Jun 26 00:09:29 1upTester Express[1929]:     at BeagleBone._pinModeInput (/var/lib/cloud9/MercuryDesign/trunk/node_modules/beaglebone-io/lib/beaglebone.js:34:10)
Jun 26 00:09:29 1upTester Express[1929]:     at BeagleBone.LinuxIO.pinMode (/var/lib/cloud9/MercuryDesign/trunk/node_modules/linux-io/lib/linux-io.js:99:12)
Jun 26 00:09:29 1upTester Express[1929]:     at Pin.set (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/pin.js:82:17)
Jun 26 00:09:29 1upTester Express[1929]:     at new Pin (/var/lib/cloud9/MercuryDesign/trunk/node_modules/johnny-five/lib/pin.js:90:13)
Jun 26 00:09:29 1upTester Express[1929]:     at Object.bb.getNewPin (/var/lib/cloud9/MercuryDesign/trunk/model/beagleboneIO.js:35:19)
Jun 26 00:09:29 1upTester Express[1929]:     at Object.pc.initialize (/var/lib/cloud9/MercuryDesign/trunk/model/PinControl.js:29:39)
Jun 26 00:09:29 1upTester systemd[1]: express.service: Main process exited, code=exited, status=1/FAILURE
Jun 26 00:09:29 1upTester systemd[1]: express.service: Unit entered failed state.
Jun 26 00:09:29 1upTester systemd[1]: express.service: Failed with result 'exit-code'.
Jun 26 00:09:31 1upTester systemd[1]: Stopped Express.
```

This same program when run via "npm start" . . . and not as a service . . . runs fine with no errrors.

So what do you make of:
```
Error: EACCES: permission denied, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
```
?

Please help!  I'm so close, but if I can't run as a service I'm hosed.

Thanks.

-roger

totem415, 2018-06-27:
ANY comment would be helpful here, even "I'm on vacation and can't deal with this now" or "I'm looking into this."  I'm starting to think that your product just won't work when run as a service, which is quite a heartbreaker given the amount of time I've spent with it.  Please say something!

fivdi, 2018-06-27:
```
Error: EACCES: permission denied, open '/sys/devices/platform/ocp//ocp:P9_23_pinmux/state'
```
This EACCES error indicates that the user does not have the privileges required to access `/sys/devices/platform/ocp/ocp:P9_23_pinmux/state`. You'll need to figure out why the user doesn't have the required privileges. I'm going to go ahead and close this issue as it's not related to BeagleBone-IO.

closed 2018-06-27 by fivdi

totem415, 2018-06-27:
@fivdi thank you very, very much for responding.

But can you please reopen this issue?

I solved my permissions problem and I'm back to the error I had at the top of this page:
```
Jun 27 23:46:59 1upTester Express[1845]: 1530143219441 Board Closing.
```

Please help me understand how "Board Closing" is a permissions issue.  I think it's an issue with Beaglebone IO when run as a service.  I'm running as the same user I SSH in as and do all my work as.  This user is not root but doesn't need to be to launch with 'npm start' and everything works fine.

Have you guys tested this when run as a service?  There must be something different about it but I can't figure out what it is.

Please comment with anything if you have any ideas.

Thanks!

-roger

fivdi, 2018-06-28:
> Have you guys tested this when run as a service?

I haven't myself and can't provide any information related to the topic.

If it turns out that the problem is being caused by BeagleBone-IO the issue can be reopened.

totem415, 2018-06-28:
@fivdi thank you so much for responding.

:-)

Yes, let's re-open.  I'd do it but I don't have a reopen button on my page.  I must not have privileges on your page?  Not sure.

I spent some time banging on this this week and I have tried several different users.  I think it was fine to begin with when not specifying a user (defaults to root) in my express.service file, where this issue began.  When I switched to a new user w/out access privileges is when I got the "permission denied" error.  Currently I have specified the same user that I used to develop the project and I'm getting the "Board Closing" error.  I can launch the node js app "npm start" w/out "sudo" or anything like that and it all works fine.  But as a service w/ the same user specified I get "Board Closing."   This feels to me like a BeagleBone-IO error. 

I'd be happy to bang on this some more if you have any ideas.  Please, even if you just have a hunch, throw it my way and I'll bang on it and let you know what happened.  But I think this is an issue with BeagleBone-IO.

My most sincere thanks for your ongoing support,

-roger

RobertCNelson, 2018-06-28:
@totem415 it works while under user, but not under systemd.

Thus, whatever environment systemd is setting up "when" it runs your script is breaking 'your' script..

Stick some "env" etc calls in your script so you can debug what systemd is setting up..

Regards,

totem415, 2018-06-28:
@RobertCNelson, 

TYVM.

That gives me something to bang on.  Will report back.

-roger

totem415, 2018-07-03:
@fivdi I was away for a few days but I'm back and I've been thinking about this issue, which is going to scuttle my whole project if I don't figure something out soon.

Maybe there's something with the Express Service that I'm running that doesn't play well with your product.  So here's my question:  How would you launch an IOT application running on the BBB that should always be up and running when the BBB is powered?  How have you guys done it that you've tested and works with your product?  Maybe I should try that.  I'm looking for a way out of this bad place I'm in.  Migrating away from Express is going to be brutal, but I appear to be running out of options.  It seems that your product does not work when run as an Express Service.  I'd love to be wrong, but you guys aren't helping me any more, so I feel like you've given up on it.

Any comments are very appreciated.

-roger

totem415, 2018-07-03:
@RobertCNelson I just noticed your comment.

I'll try sticking env before npm start and let you know.

-rog

totem415, 2018-07-03:
@RobertCNelson here is the full output of 'env npm start', including a bunch of output from my application:

```
root@1upTester:/var/lib/cloud9/MercuryDesign/trunk# env npm start

> test1up@0.0.1 start /var/lib/cloud9/MercuryDesign/trunk
> node ./bin/www

1530639593980 Available BeagleBone-IO
socketToClient initialized to: [object Object]
2018-07-03T17:39:54.082Z - info: _____________________________________
2018-07-03T17:39:54.125Z - info: 1UP TESTER SOCKPRESS EXPRESS NodeJS SERVER
2018-07-03T17:39:54.133Z - info: Initializing the testing system...
2018-07-03T17:40:02.372Z - info: app.js setup complete, connecting to hardware.
=========================== port: 8888
1530639602688 Connected BeagleBone-IO
1530639602800 Repl Initialized
>> board is ready.  Initializing LED Ctrl
______________________________deviceMgr: executeSetup
process.env: [object Object]
SUDO_GID: 1000
npm_package_dependencies_serve_favicon: ~2.4.2
npm_config_cache_lock_stale: 60000
npm_config_legacy_bundling:
npm_config_sign_git_tag:
MAIL: /var/mail/root
USER: root
npm_config_user_agent: npm/3.10.10 node/v6.13.0 linux arm
npm_config_always_auth:
npm_package_dependencies_socket_io: ~1.7.3
npm_config_bin_links: true
npm_config_key:
npm_config_description: true
npm_config_fetch_retries: 2
npm_config_heading: npm
npm_config_if_present:
npm_config_init_version: 1.0.0
npm_config_user: 1000
npm_node_execpath: /usr/bin/nodejs
SHLVL: 1
HOME: /root
OLDPWD: /home/debian
npm_config_force:
npm_config_only:
npm_config_cache_min: 10
npm_config_init_license: ISC
npm_package_dependencies_node_svn_ultimate: ^1.1.0
npm_config_editor: vi
npm_config_rollback: true
npm_config_tag_version_prefix: v
npm_config_cache_max: Infinity
npm_config_userconfig: /root/.npmrc
npm_package_dependencies_moment: ^2.18.1
npm_config_engine_strict:
npm_config_init_author_name:
npm_config_init_author_url:
npm_config_tmp: /tmp
npm_package_description: Mercury Design Testing Application
npm_config_depth: Infinity
npm_config_save_dev:
npm_config_usage:
npm_config_cafile:
npm_config_progress: true
npm_config_https_proxy:
SUDO_UID: 1000
npm_package_dependencies_async: ^2.4.1
npm_config_onload_script:
TMPDIR: /var/lib/cloud9/MercuryDesign/trunk
LOGNAME: root
npm_config_rebuild_bundle: true
[... 120 lines cut ...]
deviceMgr: led control initialized.  Success: true
uutPresenceDetect: initialize
beagleboneIO: creating new pin: P9_41
setting pin.  opts.addr: undefined opts.pin: P9_41
beagleboneIO: creating new pin: P9_27
setting pin.  opts.addr: undefined opts.pin: P9_27
deviceMgr:  uut presence detection initialized. success: true
Pin: LOW
uutPresenceDetect: Presence Pin went LOW
uutPresenceDetect: Lid Pin went LOW
>> ledControl LED OFF at: P8_7
ledControl LED OFF at: P8_9
ledControl LED OFF at: P8_11
ledControl LED OFF at: P8_8
ledControl LED OFF at: P8_10
ledControl LED OFF at: P8_12
ledControl LED OFF at: P8_14
ledControl LED OFF at: P8_16
ledControl LED OFF at: P8_18
ledControl LED OFF at: P8_15
```

Thanks again for your help!  Any thoughts are greatly appreciated.

-roger

totem415, 2018-07-03:
Hi @RobertCNelson.

Here is the complete output of 'env npm start' including a bunch of output from my application itself.
```
debian@1upTester:/var/lib/cloud9/MercuryDesign/trunk$ env npm start

> test1up@0.0.1 start /var/lib/cloud9/MercuryDesign/trunk
> node ./bin/www

1530639796854 Available BeagleBone-IO
socketToClient initialized to: [object Object]
2018-07-03T17:43:16.952Z - info: _____________________________________
2018-07-03T17:43:16.995Z - info: 1UP TESTER SOCKPRESS EXPRESS NodeJS SERVER
2018-07-03T17:43:17.002Z - info: Initializing the testing system...
2018-07-03T17:43:20.755Z - info: app.js setup complete, connecting to hardware.
=========================== port: 8888
1530639800871 Connected BeagleBone-IO
1530639800923 Repl Initialized
>> board is ready.  Initializing LED Ctrl
______________________________deviceMgr: executeSetup
process.env: [object Object]
npm_package_dependencies_serve_favicon: ~2.4.2
npm_config_cache_lock_stale: 60000
npm_config_legacy_bundling:
npm_config_sign_git_tag:
MAIL: /var/mail/debian
USER: debian
SSH_CLIENT: 192.168.2.1 51901 22
npm_config_user_agent: npm/3.10.10 node/v6.13.0 linux arm
npm_config_always_auth:
npm_package_dependencies_socket_io: ~1.7.3
npm_config_bin_links: true
npm_config_key:
npm_config_description: true
npm_config_fetch_retries: 2
npm_config_heading: npm
npm_config_if_present:
npm_config_init_version: 1.0.0
npm_config_user:
npm_node_execpath: /usr/bin/nodejs
SHLVL: 1
OLDPWD: /home/debian
HOME: /home/debian
npm_config_force:
SSH_TTY: /dev/pts/1
npm_config_only:
npm_config_cache_min: 10
npm_config_init_license: ISC
npm_package_dependencies_node_svn_ultimate: ^1.1.0
npm_config_editor: vi
npm_config_rollback: true
npm_config_tag_version_prefix: v
npm_config_cache_max: Infinity
npm_config_userconfig: /home/debian/.npmrc
npm_package_dependencies_moment: ^2.18.1
npm_config_engine_strict:
npm_config_init_author_name:
npm_config_init_author_url:
npm_config_tmp: /tmp
npm_package_description: Mercury Design Testing Application
npm_config_depth: Infinity
npm_config_save_dev:
npm_config_usage:
npm_config_cafile:
npm_config_progress: true
npm_config_https_proxy:
npm_package_dependencies_async: ^2.4.1
npm_config_onload_script:
LOGNAME: debian
npm_config_rebuild_bundle: true
[... 121 lines cut ...]
deviceMgr: led control initialized.  Success: true
uutPresenceDetect: initialize
beagleboneIO: creating new pin: P9_41
setting pin.  opts.addr: undefined opts.pin: P9_41
beagleboneIO: creating new pin: P9_27
setting pin.  opts.addr: undefined opts.pin: P9_27
deviceMgr:  uut presence detection initialized. success: true
Pin: LOW
uutPresenceDetect: Presence Pin went LOW
uutPresenceDetect: Lid Pin went LOW
ledControl LED OFF at: P8_7
ledControl LED OFF at: P8_9
ledControl LED OFF at: P8_11
ledControl LED OFF at: P8_8
ledControl LED OFF at: P8_10
ledControl LED OFF at: P8_12
ledControl LED OFF at: P8_14
ledControl LED OFF at: P8_16
ledControl LED OFF at: P8_18
ledControl LED OFF at: P8_15
```

Thanks again for your help.  Any of your thoughts are very welcome.

-roger

fivdi, 2018-07-03:
> Maybe there's something with the Express Service that I'm running that doesn't play well with your product.

Perhaps, but it's not very likely.

> So here's my question: How would you launch an IOT application running on the BBB that should always be up and running when the BBB is powered? How have you guys done it that you've tested and works with your product?

As previously mentioned [here](https://github.com/julianduque/beaglebone-io/issues/73#issuecomment-401030985) this isn't something that I tried in the past so I read [this](https://nodesource.com/blog/running-your-node-js-app-with-systemd-part-1/).

Here's a simple Johnny-Five/BeagleBone-IO application that blinks two of the on-board LEDs on a BeagleBone Black. Note how the application sets repl to false to turn off the REPL (Read Eval Print Loop). It's essential to turn off the REPL in a service.

```js
debian@beaglebone:~/beaglebone-io$ pwd
/home/debian/beaglebone-io
debian@beaglebone:~/beaglebone-io$ cat built-in-leds.js 
'use strict';

var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
  io: new BeagleBone(),
  repl: false // IMPORTANT: SET REPL TO FALSE HERE
});

board.on('ready', function() {
  var led2 = new five.Led('USR2'),
    led3 = new five.Led('USR3');

  led2.blink(500);
  led3.blink(500);
});
```

Here's the content of `/lib/systemd/system/beaglebone_io.service`:

```
[Unit]
Description=built-in-leds.js - blink onboard leds
After=network.target

[Service]
Type=simple
User=debian
ExecStart=/usr/bin/node /home/debian/beaglebone-io/built-in-leds.js
Restart=on-failure

[Install]
WantedBy=multi-user.target
```
To start the application when the machine boots run:
```
sudo systemtl enable beaglebone_io
```
When the BeagleBobe Black boots, the Johnny-Five/BeagleBone-IO application starts and the on-board LEDs blink. Everything works as expected.

totem415, 2018-07-03:
Wonderful, @fivdi .

That gives me something to try... and I'm glad to hear that you don't think it's Express' fault.

Will report back.

Thanks!

-roger

totem415, 2018-07-04:
Money!!!

@fivdi you are very, very good.  If someone tells you otherwise, you just refer them to me.  Thank you so much!!!

I had one other call that caused an error after I disabled repl, but I just commented it out and it doesn't seem to break my app:
```
i2cDevices[accessID].board.repl.inject({ replWrite });
```

I am very stoked that you solved this for me.  Thanks a million.

:-)

-roger