julianduque/beaglebone-io #74: Edit the PWM frequency at the Raspberry?

Ch3fk0ch88, 2018-09-24:
Hi,
is there a way the change the pwm frequenzy in the "beaglebone.js" file at the Raspberry?
Where i can find this file?

fivdi, 2018-09-26:
beaglebone-io is only for usage on the BeagleBone Black, BeagleBone Green Wireless and PocketBeagle. It's not intended for usage on Raspberry Pi. It's a JohnnyFive IO plugin and provides an API for modifying the PWM duty-cycle but not the PWM frequency.

Ch3fk0ch88, 2018-09-26:
But i found an beaglebone.js on the Raspberry, but an Edit dont change the frequency.

fivdi, 2018-09-26:
> But i found an beaglebone.js on the Raspberry, but an Edit dont change the frequency.

Maybe you installed it with the command `npm install beaglebone-io`?

Are you sure you have a Raspberry Pi and not a BeagleBone?
Why do you call it a Raspberry and not a Raspberry Pi?

Ch3fk0ch88, 2018-09-26:
Yes, i installed it and yes i use an Pi 3b+.

@ the "node-red-contrib-gpio" i found this:
https://github.com/monteslu/node-red-contrib-gpio/issues/31#issuecomment-373880977

An here i found this:
https://github.com/julianduque/beaglebone-io/issues/65#issuecomment-374375382

But a change of the value dont change the frequenzy

fivdi, 2018-09-26:
Ok, I think I understand now.

The node-red-contrib-gpio comment linked to above (https://github.com/monteslu/node-red-contrib-gpio/issues/31#issuecomment-373880977) mentions beaglebone-io because the first post in the issue is from someone who has a BeagleBone Black. The issue was about a BeagleBone Black and not about a Raspberry Pi.

beaglebone-io does not work on a Raspberry Pi. There is no point in changing anything in beaglebone.js as it will not result in any change to the PWM frequency on a Raspberry Pi.

There are libraries that can be used to control PWM frequency on a Raspberry Pi, for example, [pigpio](https://github.com/fivdi/pigpio). Take a look at the [hardwarePwmWrite(frequency, dutyCycle) method](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#hardwarepwmwritefrequency-dutycycle). It should be possible to achieve what you would like to achieve with this method.

Ch3fk0ch88, 2018-09-26:
Ahhh u are right sorry for that.

fivdi, 2018-09-26:
No problem :smile:

closed 2018-09-26 by fivdi