julianduque/beaglebone-io #78: Typo in README

Kal9Isa, 2020-12-13:
Regarding pin definition in the code for output/input modes:

`board.on('ready', function () {
  this.pinMode('USR3', this.MODES.OUTPUT);
  this.digitalWrite('USR3', this.HIGH);
  this.pinMode('A0', this.MODES.ANALOG);
  this.analogRead('A0', function (value) {
    console.log(value);
  });
});`

It should be this.MODE.OUTPUT instead of MODES.

closed 2020-12-13 by Kal9Isa

Kal9Isa, 2020-12-13:
I'm having an issue with the following response:

`
this.pinMode('USR3',this.MODE.OUTPUT);
TypeError: Cannot read property 'OUTPUT' of undefined
`

And can't reproduce the resolution using `MODE` instead of `MODES`.

fivdi, 2020-12-13:
The test program in section ["Standalone Usage of BeagleBone-IO"](https://github.com/julianduque/beaglebone-io#standalone-usage-of-beaglebone-io) should run without error. The correct way to specify the mode is `this.MODES.OUTPUT`, not `this.MODE.OUTPUT`.

Please post a complete program that can be used to reproduce the error that you are seeing.